microsoft/Azure-Kinect-Sensor-SDK #1720: pre-compiled 3d viewer not working & sample not working

nirvanaczj, 2021-12-27:
**Describe the bug**
In BT version 1.1.0, The pre-compiled 3D viewer will close after several seconds with GPU mode without showing anything. In BT version 1.0.1, it will hang without showing anything. 

Moreover, if I tried to build my own executable from the sample, the sample freezes on the function `k4abt_tracker_create()` function as same as for issue #1651.

When I tried to run on CPU mode, it returned 

> global variable gl_FragColor is removed after version 420
 
Moreover, the 3d viewer worked for me when I didn't update my CUDA. However, the sample also froze on the function `k4abt_tracker_create()` function before the CUDA update.

**Expected behavior**

**Logs**

```

> [2021-12-27 22:06:13.116] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (53): k4a_device_t_create(). Created   k4a_device_t 0000029743890C80
> [2021-12-27 22:06:13.116] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth_mcu\depth_mcu.c (29): depthmcu_t_create(). Created   depthmcu_t 00000297438B3DC0
> [2021-12-27 22:06:13.116] [trace] [t=18376] d:\a\1\s\extern\azure-kinect-sensor-sdk\src\usbcommand\usb_cmd_priv.h (101): usbcmd_t_create(). Created   usbcmd_t 000002974387C4D0
> [2021-12-27 22:06:13.127] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (337): find_libusb_device(). Container ID found: {e27a34e6-2fba-4a2b-4684-8c20f40ab2d6} 
> [2021-12-27 22:06:13.127] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 000118414812
> [2021-12-27 22:06:13.127] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (647): usb_cmd_io(). XFR: Cmd=00000115, PayloadSize=255
> [2021-12-27 22:06:13.141] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color_mcu\color_mcu.c (41): colormcu_t_create(). Created   colormcu_t 00000297438BE770
> [2021-12-27 22:06:13.141] [trace] [t=18376] d:\a\1\s\extern\azure-kinect-sensor-sdk\src\usbcommand\usb_cmd_priv.h (101): usbcmd_t_create(). Created   usbcmd_t 00000297438C6DF0
> [2021-12-27 22:06:13.149] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (337): find_libusb_device(). Container ID found: {e27a34e6-2fba-4a2b-4684-8c20f40ab2d6} 
> [2021-12-27 22:06:13.149] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 000118414812
> [2021-12-27 22:06:13.149] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\calibration\calibration.c (46): calibration_t_create(). Created   calibration_t 00000297438DF1B0
> [2021-12-27 22:06:13.149] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (647): usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=10240
> [2021-12-27 22:06:13.171] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (54): capturesync_t_create(). Created   capturesync_t 00000297438DEF00
> [2021-12-27 22:06:13.171] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\queue\queue.c (39): queue_t_create(). Created   queue_t 0000029743909700
> [2021-12-27 22:06:13.171] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\queue\queue.c (39): queue_t_create(). Created   queue_t 00000297439097A0
> [2021-12-27 22:06:13.171] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\queue\queue.c (39): queue_t_create(). Created   queue_t 00000297439096B0
> [2021-12-27 22:06:13.171] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (54): depth_t_create(). Created   depth_t 00000297438C2780
> [2021-12-27 22:06:13.171] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (647): usb_cmd_io(). XFR: Cmd=00000201, PayloadSize=18
> [2021-12-27 22:06:13.175] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (647): usb_cmd_io(). XFR: Cmd=00000201, PayloadSize=18
> [2021-12-27 22:06:13.180] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (278): ******************** Device Info ********************
> [2021-12-27 22:06:13.180] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (279): K4A SDK version:     1.3.0
> [2021-12-27 22:06:13.180] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (647): usb_cmd_io(). XFR: Cmd=00000115, PayloadSize=255
> [2021-12-27 22:06:13.206] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (285): Serial Number:       000118414812
> [2021-12-27 22:06:13.206] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (289): RGB Sensor Version:  1.6.110
> [2021-12-27 22:06:13.207] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (292): Depth Sensor Version:1.6.80
> [2021-12-27 22:06:13.207] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (295): Mic Array Version:   1.6.14
> [2021-12-27 22:06:13.207] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (298): Sensor Config:       6109.7
> [2021-12-27 22:06:13.207] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (301): Build type:          Release
> [2021-12-27 22:06:13.207] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (306): Signature type:      MSFT
> [2021-12-27 22:06:13.207] [critical] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (312): ****************************************************
> [2021-12-27 22:06:13.207] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dewrapper\dewrapper.c (57): dewrapper_t_create(). Created   dewrapper_t 00000297438A8D70
> [2021-12-27 22:06:13.207] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\queue\queue.c (39): queue_t_create(). Created   queue_t 0000029743909B60
> [2021-12-27 22:06:13.207] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (647): usb_cmd_io(). XFR: Cmd=000000f2, PayloadSize=0
> [2021-12-27 22:06:13.223] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (647): usb_cmd_io(). XFR: Cmd=0000000a, PayloadSize=0
> [2021-12-27 22:06:13.236] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color\color.cpp (45): color_t_create(). Created   color_t 00000297438C4FD0
> [2021-12-27 22:06:13.257] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\imu\imu.c (70): imu_t_create(). Created   imu_t 0000029743953830
> [2021-12-27 22:06:13.257] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\queue\queue.c (39): queue_t_create(). Created   queue_t 00000297438DCF60
> [2021-12-27 22:06:13.257] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (647): usb_cmd_io(). XFR: Cmd=80000004, PayloadSize=0
> [2021-12-27 22:06:13.257] [trace] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (846): k4a_device_start_cameras(). k4a_device_start_cameras starting
> [2021-12-27 22:06:13.257] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (864): k4a_device_start_cameras(). Starting camera's with the following config.
> [2021-12-27 22:06:13.257] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (865): k4a_device_start_cameras().     color_format:0
> [2021-12-27 22:06:13.257] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (866): k4a_device_start_cameras().     color_resolution:0
> [2021-12-27 22:06:13.257] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (867): k4a_device_start_cameras().     depth_mode:2
> [2021-12-27 22:06:13.257] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (868): k4a_device_start_cameras().     camera_fps:2
> [2021-12-27 22:06:13.257] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (869): k4a_device_start_cameras().     synchronized_images_only:0
> [2021-12-27 22:06:13.257] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (870): k4a_device_start_cameras().     depth_delay_off_color_usec:0
> [2021-12-27 22:06:13.257] [info] [t=18376] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (871): k4a_device_start_cameras().     wired_sync_mode:0
[... cut ...]
> [2021-12-27 22:06:13.499] [trace] [t=19972] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\allocator\allocator.c (118): k4a_capture_t_create(). Created   k4a_capture_t 0000029751949110
> [2021-12-27 22:06:13.499] [trace] [t=19972] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (51): k4a_image_t_create(). Created   k4a_image_t 00000297518F31A0

```

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Windows 11 (GPU: RTX 3060)
 - SDK Version:  Sensor SDK 1.4.1 with BT  1.1.0  (also tested on multiple other versions (Sensor SDK 1.3.0 & BT 1.0.1 & 1.0.0))
 - Firmware version: 
 - 
**Additional context**

nirvanaczj, 2021-12-28:
update:

I changed my graphics card driver from 497.29 to 472.84 and now the 3d viewer works.

Moreover, I copied all the files under the bt_sdk/tool/ without the 3d viewer to my sample directory, now the sample works. Then I did some cross-validation(only copying one file at a time) to see which file triggered the problem.

The result is that:

only my executable file + all other files under tools => OK
only my executable file + my onnxruntime.dll + all other files under tools => not working:` Body tracker initialization failed!`
all my sample project files excluding onnxruntime.dll + onnxruntim.dll under tools folder => not working: `Body tracker initialization failed!
`

the onnxruntime within the body tracking 1.10 might have some compatibility issues.

fkavache, 2022-01-05:
Hi @nirvanaczj 

The problem is that Nvidia drivers, If I’m not mistaken, after the 496.13 version update (Oct-12) are having some issues with OpenGL, that's why 472.84 is working fine.

JamesMcCullochDickens, 2022-01-17:
@nirvanaczj 
What version of CUDA and nvidia driver did you use to make the k4abt_tracker_create function stop hanging?

frabera, 2022-04-01:
Is there any update? I have the same problem

fkavache, 2022-04-07:
Hi, please take a look at this thread.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1763#issuecomment-1091234984

qm13, 2022-05-25:
Fixed in v1.1.2

closed 2022-05-25 by qm13