microsoft/Azure-Kinect-Sensor-SDK #1721: How to configure CUDA mode in VS（it can run in cmd）

qyupupup, 2021-12-30:
**Describe the bug**
Hello, I can run body tracking in CMD to go through CUDA mode. (it displays an error, but Microsoft officials say it is safe), but how can I configure it in vs2017?  Because the configuration function in k4abtttypes. h only has mode_gpu and mode_cpu,and hasn't K4ABT_TRACKER_PROCESSING_MODE_GPU_CUDA(I see [in](https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#specifying-onnx-runtime-execution-environment) . 
If I select CPU, the FPS of body tracking will become slow, but when I select mode_gpu, an inoperable blank window will appear after compilation. 

**To Reproduce**

body tracking can be run in cmd through CUDA( FPS high )、cpu( FPS mid ）、directml( FPS slow to about 1) and TensorRT( appear a inoperable blank window ). 
and k4abt_tracker_processing_mode_t has only two modes, MODE_GPU and MODE_CPU
**Expected behavior**

I can configure CUDA mode in vs2017, or how to made the GPU mode can run

**Screenshots**

[image]
[image]
**Desktop (please complete the following information):**
 - CPU : intel i7-11800H 
 - video card : nvidia rtx 3050 laptop & driver472.19 & cuda 11.1 & cudnn8.0.5
 - OS with Version:  Win11 home Chinese version
 - SDK Version:   Kinect 1.4.1 & body tracking 1.1.0
 - Firmware version:     
 RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
 

**Additional context**

`typedef enum
{
    K4ABT_TRACKER_PROCESSING_MODE_GPU = 0, /**< SDK will use GPU mode to run the tracker */
    K4ABT_TRACKER_PROCESSING_MODE_CPU,     /**< SDK will use CPU only mode to run the tracker */
} k4abt_tracker_processing_mode_t;`

fkavache, 2021-12-31:
Hi @qyupupup

In the latest 1.1.0 version of the BT SDK there are additional options for CUDA, TENSORRT and DIRECTML

```C
typedef enum
{
    K4ABT_TRACKER_PROCESSING_MODE_GPU = 0, /**< SDK will use the most appropriate GPU mode for the operating system to run the tracker */
    /**< Currently this is ONNX DirectML EP for Windows and ONNX Cuda EP for Linux. ONNX TensorRT EP is experimental */
    K4ABT_TRACKER_PROCESSING_MODE_CPU,     /**< SDK will use CPU only mode to run the tracker */
    K4ABT_TRACKER_PROCESSING_MODE_GPU_CUDA,     /**< SDK will use ONNX Cuda EP to run the tracker */
    K4ABT_TRACKER_PROCESSING_MODE_GPU_TENSORRT, /**< SDK will use ONNX TensorRT EP to run the tracker */
    K4ABT_TRACKER_PROCESSING_MODE_GPU_DIRECTML /**< SDK will use ONNX DirectML EP to run the tracker (Windows only) */
} k4abt_tracker_processing_mode_t;
```
You are running on 1.0.1

qyupupup, 2022-01-02:
Hi @fkavache 
I rechecked my library file and found that it refers to the library file in nuget package (version 1.0.1) .And I uninstalled the library file and configured it again, and now it can run normally in CUDA mode.

Thank you for bringing me the first good news of 2022, my thanks is beyond the words description

closed 2022-01-03 by qm13