microsoft/Azure-Kinect-Sensor-SDK #1722: Support for all color camera parameters in k4arecorder

borosko, 2021-12-30:
**Is your feature request related to a problem? Please describe.**
k4arecorder not able to set color camera parameters

**Describe the solution you'd like**
k4arecorder supports setting color camera parameters, specifically whiteBalance, brightness, contrast, saturation, sharpness.

**Describe alternatives you've considered**
A workaround is to set these parameters manually from k4aviewer, but this poses a significant strain and is very error prone when trying to reproduce recordings of similar quality.

**Additional context**
This is already implemented in https://github.com/borosko/Azure-Kinect-Sensor-SDK fork. I'm creating this issue to be triaged so I can create a PullRequest.
My team has been using this build successfully for the past 6 months. We've been replacing the default 1.4.1 k4arecorder using this msi: https://github.com/borosko/Azure-Kinect-Sensor-SDK/releases/tag/coloropt

borosko, 2022-03-11:
This is not getting any traction for 3+ months. Is there anything I should do here? I have prepared a PR, but just waiting for an issue to be approved to be able to create it. @qm13

qm13, 2022-03-11:
@borosko please go ahead and create a PR.