microsoft/Azure-Kinect-Sensor-SDK #1724: Azure Kinect Viewer memory leak

adahbingee, 2022-01-06:
Azure Kinect SDK 1.4.1

**Describe the bug**

memory leak with Azure Kinect Viewer

stop preview not release memory

**To Reproduce**
1. Open Azure Kinect Viewer
2. Open device and start preview

**Expected behavior**

Memory usage stable.

**Logs**

```
[ info     ] : k4a_device_start_cameras(). Starting camera's with the following config.
[ info     ] : k4a_device_start_cameras().     color_format:3
[ info     ] : k4a_device_start_cameras().     color_resolution:1
[ info     ] : k4a_device_start_cameras().     depth_mode:2
[ info     ] : k4a_device_start_cameras().     camera_fps:2
[ info     ] : k4a_device_start_cameras().     synchronized_images_only:1
[ info     ] : k4a_device_start_cameras().     depth_delay_off_color_usec:0
[ info     ] : k4a_device_start_cameras().     wired_sync_mode:0
[ info     ] : k4a_device_start_cameras().     subordinate_delay_off_master_usec:0
[ info     ] : k4a_device_start_cameras().     disable_streaming_indicator:0
[ info     ] : k4a_device_start_cameras(). k4a_device_start_cameras started
[ info     ] : capturesync_add_capture(). Dropped 5 depth captures waiting for time stamps to stabilize
[ info     ] : drop_sample(). capturesync_drop, Dropping sample TS:    200000 type:Depth
[ info     ] : drop_sample(). capturesync_drop, Dropping sample TS:    268533 type:Depth
[ info     ] : drop_sample(). capturesync_drop, Dropping sample TS:    301866 type:Depth
```

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows10 pro 21H1
 - SDK Version:  1.4.1
 - Firmware version:  
 ```
 Current Firmware Versions:
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```
**Additional context**

fkavache, 2022-01-10:
Hi @adahbingee

Could you please provide more detailed information about:
1. What is the peak of memory usage is your case.
2. How much time it needs to peak.
3. What's happening when memory is full.

adahbingee, 2022-01-11:
> Hi @adahbingee
> 
> Could you please provide more detailed information about:
> 
> 1. What is the peak of memory usage is your case.
> 2. How much time it needs to peak.
> 3. What's happening when memory is full.

Solved with lastest firmware

closed 2022-01-13 by qm13