microsoft/Azure-Kinect-Sensor-SDK #1725: Image artifacts preventing localization

kkh3049, 2022-01-06:
**Describe the bug**
We are interested in registering Kinect color images with other color camera images.  As part of this process we image an object with targets of specific metrics with both and search for matching targets between the Kinect and other color images.  As we pursue this, we found that the Kinect performs what looks like sharpening on the images, and this created artifacts of unrealistic color bands on our targets which throws off the application we utilize in this process.
See the comparison of images from different sources in the screenshots section.
Is there some way to prevent this sharpening?  A setting to change.  Or is there a way to retrieve raw color images from the Kinect, so we may process them ourselves?

**To Reproduce**
Take a color image of a black on white object with the Kinect Camera at any resolution.

**Expected behavior**
No white banding or black edging artifacts as shown in an image from another camera.

**Screenshots**
Image from the Kinect:
[image]
Image from another Camera:
[image]

Notice the faint white banding in the Kinect Image, and the lack of such banding in the other Camera's image

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 v2004, build 19041.1415
 - SDK Version:  .NET Nuget v1.4.1
 - Firmware version:  RGB Camera: 1.6.110; Depth camera: 1.6.70; Audio: 1.6.14

kkh3049, 2022-05-02:
This is still a concern that we are experiencing.  If it's not possible to get the RAW image directly (Note we only take snapshots with the Kinect, and do not need this data streamed continuously), is there a particular algorithm that's been applied (like say, a sharpening filter) that we could reverse?

kkh3049, 2022-05-05:
Is there any update on this?

kkh3049, 2022-05-06:
@dasparli ?

dasparli, 2022-05-07:
I'm looking into this now.
In the meantime, the Azure Kinect Viewer has a sharpness control for the color camera. Have you tried changing this value to see if different settings have an effect on these artifacts?

kkh3049, 2022-05-16:
@dasparli Thank you for the suggestion.  I have rerun the process with this suggestion and have some comparisons that I hope will help you in your process to correct this concern.
First, changing the sharpness did not result in the desired identification and localization of targets.  Reducing the sharpness, however, _did_ reduce the amount of this artifact occurring, though didn't prevent it entirely.
At sharpness level 0 we did get one target correctly coded, though the rest were not coded at all.  For context, we use two types of targets, one set is all the same white circle with a black border; the other is as you saw in the above images - one white circle, surrounded by a black border, than segments of a white border, and all surrounded by black.  These segments are unique for each target allowing for unique coding and successfully disambiguating camera localization.  Our problem with this sharpening is that the third party system we use is unable to correctly identify the coded targets.
Here is the only target in a sample image that was coded correctly in this system (when sharpness was set to 0):
[image]
This same target when the sharpness was set to 4 looks as follows: 
[image]
The white band as mentioned previously, is not nearly as present in either of these images between the white segments, however, the same type of artifact on the edges, this black banding, (and increased white banding on the edges of the white segment) is very apparent in the sharpness level 4 image, and still present, though much subdued in the sharpness level 0 image.
You can see further examples of this in looking at the following image of an uncoded target which is misidentified as a coded target when shot with sharpness at level 4:
[image]
As I mentioned, when shot with the lowest sharpness only one coded target (of almost 40) is identified as a coded target because of the this sharpening banding.
As a further example this is a failing coded target (the next best target in the current view) that fails at sharpness level 0:
[image]

While you continue to look into this, is there anything else you think we may be able to try to get less of this banding?
Thanks!

kkh3049, 2022-06-16:
@dasparli Any insights or movement on this?
Thanks in advance.

dasparli, 2022-06-21:
The ISP block that processes frames from the RGB camera can be configured and controlled by calling [ k4a_device_get_color_control() ](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/develop/group___functions_ga5cb97bc0db4ab28ad98c44e323da1cce.html#ga5cb97bc0db4ab28ad98c44e323da1cce)

You can see this function being used to configure default parameters [here](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicrosoft%2FAzure-Kinect-Sensor-SDK%2Fblob%2F2feb3425259bf803749065bb6d628c6c180f8e77%2Ftools%2Fk4aviewer%2Fk4adevicedockcontrol.cpp%23L194&data=05%7C01%7CDale.Sparling%40microsoft.com%7C301d021528b24bcaf8d808da530b312a%7C72f988bf86f141af91ab2d7cd011db47%7C0%7C0%7C637913604778786848%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=pz%2BD%2FDkto%2F7gUCoycWjCbEjOslsnNLU0XqU5ZVTm%2FTg%3D&reserved=0). 

Perhaps you can experiment with different values for some of these configurations in k4aviewer.exe in hopes of affecting those artifacts you are seeing.