microsoft/Azure-Kinect-Sensor-SDK #1726: Body Tracking does not work in GPU mode.

kakeru58, 2022-01-07:
I am currently writing code for body tracking in C#.
However, the tracking works in CPU mode, but not in GPU mode.
The code itself is fine as I have tested it on other PCs.
Here is the environment I am using.
Windows 10
Visual Studio 2019 professional
11th Gen Intel(R) Core(TM) i7-11800H @ 2.30 GHz 2.30 GHz
Geforce RTX 3050 Laptop
NVIDIA Driver:GEFORCE GAME READY 496.76
Cuda 10.0
Cudnn 7.5.1

Thanks for your help.

fkavache, 2022-01-12:
Hi @kakeru58 

Could you please be more specific about:
1. what's happening when you are trying to run in GPU mode.
2. which NVIDIA driver versions did you try.

qyupupup, 2022-01-13:
You can try to downgrade your NVIDIA driver version. I can run it by changing it to 472.56

JamesMcCullochDickens, 2022-01-24:
@qyupupup with a 30 series card?

kakeru58, 2022-01-25:
Thank you for your kind.

> Hi @kakeru58
> 
> Could you please be more specific about:
> 
> 1. what's happening when you are trying to run in GPU mode.
> 2. which NVIDIA driver versions did you try.

1.The program will not proceed from the location of this code.

"""
using (var tracker = Tracker.Create(calibration, trackerConfig))
"""

2. 472.12-notebook-win10-win11-64bit-international-dch-whql

> According to @qm13, the only tested CUDA version is 11.1. Perhaps you can try that one.

I tried it, but it didn't help.

JamesMcCullochDickens, 2022-01-25:
To add to this thread, I literally tried every single nVidia driver today with Cuda 11.1 with a 3090 and it hung on this function call.

MFIMichael, 2022-01-25:
I was having a similar issue with a build failing on Tracker.Create and managed to track it to a failure to load onnxruntime.dll. For me, I was able to get it working on an RTX 3070 using the 511.09 Driver after installing the vc++ 2015-2022 redistributables pack from https://docs.microsoft.com/en-US/cpp/windows/latest-supported-vc-redist?view=msvc-170

kakeru58, 2022-01-31:
> I was having a similar issue with a build failing on Tracker.Create and managed to track it to a failure to load onnxruntime.dll. For me, I was able to get it working on an RTX 3070 using the 511.09 [...]

I tried this.
But it's still not resolved.

fkavache, 2022-04-07:
Hi, please take a look at this thread.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1763#issuecomment-1091234984

qm13, 2022-05-25:
Fixed in v1.1.2

closed 2022-05-25 by qm13