microsoft/Azure-Kinect-Sensor-SDK #1727: Firmware on Azure Kinect is a more recent version than the SDK

ScatterCory, 2022-01-10:
The latest Azure Kinects to arrive to us have a firmware installed which is newer than the version provided here on the SDK GitHub. Here's a screenshot of the firmware tool comparing the version numbers of the Kinect's firmware with that in the downloaded binary:

[image]

If we have any issues with the sensors and need to factory reset them, will there be any problem rolling back these newer units to the currently available firmware?

Will the SDK be updated with this new firmware version, and if so, is there a timeframe when we can expect that?

Thanks in advance!

qm13, 2022-01-11:
Known issue that will be fixed in the next update to the Sensor SDK. In the meantime you can find the latest firmware file [here ](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/firmware).

alexhyy, 2022-04-21:
The latest firmware is put [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/firmware).

ScatterCory, 2022-05-10:
@alexhyy Thanks! Any idea when this will move over to the [usage page](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md)?