microsoft/Azure-Kinect-Sensor-SDK #1728: Unity BodyTracking Sample catching exception for background thread result = K4A_RESULT_FAILED

kvibert, 2022-01-11:
Describe the bug

unity Example stops at init, catching exception for background thread result = K4A_RESULT_FAILED

[SharedScreenshot png](https://user-images.githubusercontent.com/8499626/148872795-50730ad0-3a4d-40f2-b3ca-de01ecd2c60d.png)

To Reproduce
Execute all the steps on the readme.
open sample scene on unity and play.

Desktop (please complete the following information):

OS with Version:Windows 18362.1256
SDK Version: Azure Kinect SDK v1.4.1 / Azure Kinect Body Tracking SDK v1.1.0
Firmware version: Current Firmware Versions:
RGB camera firmware: 1.6.102
Depth camera firmware: 1.6.75
Depth config file: 6109.7
Audio firmware: 1.6.14
Build Config: Production
-Unity Version: 2019.3.0f1

Additional context
k4aviewer and k4abt_simple_3d_viewer works flawlessly.

anythingthatworks, 2022-01-25:
I'm having the same issue. in my research I found (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1600) and went through those fixes but still couldn't fix the problem

anythingthatworks, 2022-01-25:
So after a lot of experimentation I was able to solve my issues:

OS with Version: Windows 10 Home 19042.1466
SDK Version: Azure Kinect SDK v1.4.1 / **Azure Kinect Body Tracking SDK v1.0.1**
Firmware version: Current Firmware Versions:
**Unity Version: 2019.1.2f1**

I tried a couple of things to get this working, so I can't tell which solved the issue but here's the list:

1. use the exact unity version of the project
2. use an older version of the body tracking sdk (1.0.1 vs 1.1.0)
3. copy everything in the tools folder of the body tracker sdk into the root of the unity project (same place as the assets folder)
4. Change ProcessingMode = TrackerProcessingMode.GPU to ProcessingMode = TrackerProcessingMode.Cuda (https://github.com/microsoft/Azure-Kinect-Samples/blob/master/body-tracking-samples/sample_unity_bodytracking/Assets/Scripts/SkeletalTrackingProvider.cs#L46)

svenneve, 2022-02-08:
> 3\. copy everything in the tools folder of the body tracker sdk into the root of the unity project (same place as the assets folder)

Thanks, that's what fixed it for me. Updated to 2019.4.19f with SDK 1.1.0 and still working.

DouglasAllenDev, 2022-02-17:
Has anyone managed to get this working in Unity 2020?
I've tried copying over everything from the tools folder to the project root and changing the processing mode to Cuda but Im still getting the annoyingly vague K4A_RESULT_FAILED error. 

I really wish it would just say what the problem library is, this is not the first time Ive had to waste hours trying to debug this pointlessly vague error.

yama7575, 2022-02-21:
Please forgive my poor English because I am Japanese.

We have confirmed the operation on Unity 2020 with a similar procedure.
However, only when the graphic board is RTX2070Super, if you perform the same procedure with RTX3080, "K4A_RESULT_FAILED error." Will appear.
I'm wondering if it doesn't work due to the GPU, but is there anyone who has confirmed the operation with the RTX30 ** series?

ChiefBreakeverything, 2022-03-23:
I'm also getting this error on a 2070 Super. Kinect itself is functioning fine, applied the patch commits, etc...

Azure Kinect SDK 1.4.1
Body Tracking SDK 1.1.0

@corriedotdev when you say you're using 1.1.0 (the one that isn't on nuget) are you talking about the Azure Kinect SDK or the Body Tracking SDK? I've tried 1.1.0 and 1.1.1 of the Body Tracking SDK and neither worked. I've copied all the files to the project root and still no luck.

Vandor1, 2022-03-24:
I couldnt fix this error using [Body Tracking SDK 1.1.1](https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/1.1.1).

I resolved my issue by uninstalling this version, and instead downloaded the [1.0.1 version](https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/1.0.1). I then copied all the files from "tools" to my unity program folder. Hope this helps someone.

ChiefBreakeverything, 2022-04-03:
Amazing! I got it working by copying everything across, very naive approach, however it only works in Unity 2020. 2021 will complain about netstandard, I think whats going on is that Unity include their own version of this library for 2021 onwards?

Either way, I'll follow your tutorial because having it work in 2021 would be even better. Much appreciated!

corriedotdev, 2022-06-09:
.. for convenience https://github.com/microsoft/Azure-Kinect-Samples/commits/master

closed 2022-06-13 by qm13

fangshangping, 2024-08-10:
Through my test, first check the version of Kinect.BodyTracking in the Packages folder, mine is 1.1.2, and the original SDK of my computer is 1.0.1, so I still get an error after putting the dll in Tools into Unity. I reinstalled the SDK for 1.1.2 and put the dll from Tools into unity and it ran successfully.
通过我的测试，先查看Packages文件夹下Kinect.BodyTracking的版本，我的是1.1.2，我电脑原来的SDK是1.0.1，所以我把Tools里的dll放入到Unity中后还是报错了。我重新安装了1.1.2的SDK后再把Tools里的dll放入unity中就成功运行了。