microsoft/Azure-Kinect-Sensor-SDK #1731: Azure Kinect BodyTrackingSamples & Unity | multiple skeletons

leGITplease, 2022-01-18:
Hi guys,

Currently i have 1 skeleton tracking myself as i stand in front of the Azure Kinect, what i want to achieve is having multiple skeletons that each represent 1 user in Unity:

[image]

For instance, if i stand with another person in front of the camera i want the 2nd skeleton to track their body so you have 2 different skeletons with each their own movements linked to each single user.

Im using this repo: https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples 

Any ideas on how to do that?

KanGeQiu, 2022-07-23:
Hi, have you solved this problem, I have the same problem now, if you have solved this problem can you tell me how to solve it.

Thaina, 2022-10-27:
You can modified `SkeletalTrackingProvider.cs` directly. It was just give you only one skeleton even the device actually give many skeleton bodies

hemmkim, 2023-09-30:
> You can modified `SkeletalTrackingProvider.cs` directly. It was just give you only one skeleton even the device actually give many skeleton bodies

Hello, I was trying to modify  `SkeletalTrackingProvider.cs` but could not solve how to fix it.
Could you please help me how to modify the script?

I'd very much appreciate it if you can help!

Thaina, 2023-10-11:
@hemmkim This is too long and I mostly forget what I have done. There are many file I have modified from the original project

But if it might be useful, this is the current `SkeletalTrackingProvider.cs` in my project

```C#
using Microsoft.Azure.Kinect.BodyTracking;
using Microsoft.Azure.Kinect.Sensor;
using System;
using System.IO;
using System.Runtime.InteropServices;
using System.Threading;
using System.Threading.Tasks;
using Unity.Mathematics;
using UnityEngine;
using UnityEngine.SceneManagement;

public class SkeletalTrackingProvider : BackgroundDataProvider
{
    bool readFirstFrame = false;
    TimeSpan initialTimestamp;

    public float3 accelSample;
    public SkeletalTrackingProvider(int id) : base(id)
    {
        Debug.Log("in the skeleton provider constructor");
    }

    System.Runtime.Serialization.Formatters.Binary.BinaryFormatter binaryFormatter { get; set; } = new System.Runtime.Serialization.Formatters.Binary.BinaryFormatter();

    public Stream RawDataLoggingFile = null;
    public SensorOrientation sensorOrientation = SensorOrientation.Default;
    public TrackerProcessingMode trackerProcessingMode = TrackerProcessingMode.Cuda;
    protected override void RunBackgroundThreadAsync(int id, CancellationToken token)
    {
        try
        {
            UnityEngine.Debug.Log("Starting body tracker background thread.");
            DebugTime.isDetectCamera = true;
            // Buffer allocations.
            var currentFrameData = new BackgroundData();
            // Open device.
            using (var device = Device.Open(id))
            {
                device.StartCameras(new DeviceConfiguration() {
                    CameraFPS = FPS.FPS15,
                    ColorResolution = ColorResolution.Off,
                    DepthMode = DepthMode.NFOV_Unbinned,
                    WiredSyncMode = WiredSyncMode.Standalone,
                });
                
                Task.Run(() => {
                    device.StartImu();
                    try
                    {
                        while(!token.IsCancellationRequested)
                        {
                            try
                            {
                                Thread.Sleep(100);
                                var imu = device.GetImuSample(TimeSpan.FromMilliseconds(100));
                                accelSample = new float3(imu.AccelerometerSample.FromKinectToUnity()).yzx;
                            }
                            catch(Exception e)
                            {
                                Debug.LogException(e);
                            }
                        }
                    }
                    catch(Exception e)
                    {
[... 77 lines cut ...]
                                    // Update data variable that is being read in the UI thread.
                                    SetCurrentFrameData(ref currentFrameData);
                                }
                            }
                        }

                        Debug.Log("dispose of tracker now!!!!!");
                    }
                }
            }

            if (RawDataLoggingFile != null)
            {
                RawDataLoggingFile.Close();
            }            
        }
        catch (Exception e)
        {
            DebugTime.isDetectCamera = false;
            Debug.LogException(e);
            token.ThrowIfCancellationRequested();                        
        }
    }  
}
```

hemmkim, 2023-10-16:
Hello, 

Thank you so much for your reply.
I very much appreciate it.

I'll try this on my project again soon.

Best wishes,
HyeMin

----
김혜민 (HyeMin Kim)
+82) 10-7471-7343

@fruityhye
@higos_jardin
@orange.planet_