microsoft/Azure-Kinect-Sensor-SDK #1733: k4abt_simple_3d_viewer not working in Ubuntu 18.04 with RTX3070 with CUDA 11.1 and Cudnn 8.0.4

AswinkarthikeyenAK, 2022-01-22:
Hi, 
I am using **Ubuntu 18.04** with **Nvidia RTX 3070**. It has an Intel i7.
**Nvidia Driver Version** - 470.86
**CUDA Version** - 11.1
`~$ which nvcc`
`/usr/local/cuda-11.1/bin/nvcc`
**Cudnn** - 8.0.4

The k4aviewer works fine. But when launching the `~$ k4abt_simple_3d_viewer`, there is no response for a very long time. After which I can view the body but not the skeleton of the body with the following error in the terminal.
`[E:onnxruntime:, sequential_executor.cc:165 Execute] Non-zero status code returned while running Pad node. Name:'workaround_cudnn' Status Message: 
Non-zero status code returned while running Pad node. Name:'workaround_cudnn' Status Message: 
2022-01-22 18:24:17.756075539 [E:onnxruntime:K4ABT_ORT_Environment, cuda_call.cc:93 CudaCall] CUDA failure 48: no kernel image is available for execution on the device ; GPU=0 ; hostname=LAMS ; expr=cudaGetLastError(); 
`

Has any experienced this before? If so, can you kindly suggest me what to do next to solve this.

[image]

lewkoo, 2022-01-24:
Try installing a 472.XX Nvidia Driver version. It seems like Body Tracking was tested only against that particular version and CUDA@11.1 (but you already have that installed)

fkavache, 2022-05-31:
Hi @AswinkarthikeyenAK

Please check the latest v1.1.2 BTSDK

And make sure that you are meeting all the requirements.
https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-setup