microsoft/Azure-Kinect-Sensor-SDK #1734: Calling k4abt_tracker_create() has undesired behavior with an RTX 3090

JamesMcCullochDickens, 2022-01-22:
**Description**
Using the kabt_tracker_create() function, as used in the offline processor code [here](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/offline_processor). Running on an RTX 3090, the code stalls (up to 10 mintes!
), with the GPU enabled i.e.tracker_config.processing_mode = flag enabled (K4ABT_TRACKER_PROCESSING_MODE_GPU). After stalling, the tracker returns no bodies in the resulting json file. This behavior does not happen when using a 1070 GPU with the exact same environment details.

To Reproduce:
Run the offline processor with an RTX 3090 vs. 1070 gpu, observe the stall with a debugger when the kabt_tracker_create() is called as well as the output json having no bodies on an input with people that can be seen in the event viewer (where joints are visible in the playback of a .mkv file).

 - OS: Windows 10 10.0.19043 Build 19043
 - Body Tracking SDK Version:  1.1.0
 - nVidia Driver: 456.71

qm13, 2022-01-24:
Please upgrade to nVidia Driver to at least 460. We are seeing good results with 472.12.

JamesMcCullochDickens, 2022-01-24:
@qm13, is Cuda 11.1 ok ?

JamesMcCullochDickens, 2022-01-24:
I just tried driver 472.12, Cuda 11.1 and same issue occurred. Would be happy to give more info.