microsoft/Azure-Kinect-Sensor-SDK #1735: Unable to run Body tracking on NVIDIA Quadro RTX A4000

marsarCV, 2022-01-27:
**Describe the bug**
Hi, I have been trying to run a custom application that utilizes body tracking SDK as part of detecting human movements. The code was written last March and it ran fine on both NVIDIA Quadro RTX 4000 and NVIDIA Quadro RTX 5000. Since last week I have been trying to run the same code on NVIDIA Quadro RTX A4000 and it does not work! Here are some of the findings:

1. Program is stuck at this method: `k4abt_tracker_create(const k4a_calibration_t* sensor_calibration, k4abt_tracker_configuiration_t config, k4abt_tracker_t* tracker_handle)`
2. If I leave it running, the visualization window opens up after about 15 minutes, showing only the IR camera image without any skeletal tracking.
3. If I switch to CPU mode (like this : `tracker_config.processing_mode = inputSettings.CpuOnlyMode ? K4ABT_TRACKER_PROCESSING_MODE_CPU : K4ABT_TRACKER_PROCESSING_MODE_GPU;`) before calling k4abt_tracker_create(), I get errors and no visualization window or tracking.

Interestingly, it works as expected on NVIDIA Quadro RTX 4000 and NVIDIA Quadro RTX 5000. Driver versions tested for each card :
1. NVIDIA Quadro RTX A4000: 471.41, 471.68, 472.84, 511.09
2. NVIDIA Quadro RTX 4000: 471.41
3. NVIDIA Quadro RTX 5000: 471.47

(System: Windows 10 Pro, Azure sdk : 1.3.0, 1.4.1)

marsarCV, 2022-02-03:
Hi, my system only consists of NVIDIA graphic cards (2 x A4000). Here is the screenshot of control panel settings:

[image]

fkavache, 2022-04-07:
Hi @marsarCV
could you please take a look at this thread?

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1763#issuecomment-1091234984

marsarCV, 2022-04-29:
Hi @fkavache I tried the given executable. The sdk I am using (1.1.1) had k4abt_simple_3d_viewer.exe which was working fine already. I replaced it with the linked exe and that worked the same. My program which depends on this sdk however still behaves the same.

qm13, 2022-05-25:
Fixed in v1.1.2

closed 2022-05-25 by qm13