microsoft/Azure-Kinect-Sensor-SDK #1737: Synchro between camera only with depth stream

fdechaumont, 2022-02-01:
Dear Kinect Devs,

Thanks for your amazing device, amazing driver.
I am using your camera for research purposes ( http://livemousetracker.org )

I only use depth, but to sync them all, I need to grab the color stream that I just drop.
I guess you need color stream because it is hardware related ? Would it still be possible no to send the data over the USB to gain bandwidth ? Would this make sense if I use a lot of cameras at once for instance ?

Best

Fabrice

qm13, 2022-02-10:
The internal hardware sync is generated by the RGB camera circuitry. The master camera of a set of synchronized cameras requires the RGB camera. The subordinate cameras should not require the RGB camera.

fdechaumont, 2022-02-12:
Thanks a lot !

closed 2022-02-12 by fdechaumont