microsoft/Azure-Kinect-Sensor-SDK #1739: Fail to record using Azure Kinect DK recorder

pythonsql21, 2022-02-04:
I am new to Azure Kinect. After installing the MSI package of azure Kinect SDK (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md), I tried to record data stream using Azure Kinect DK recorder but got an error like below.
[image]

I used Firmware Versions as below.
  OS: Windows 10
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.80
  Depth config file:        6109.7
  Audio firmware:           1.6.14
I could use Azure Kinect viewer v1.4.1 without any problem but failed to record the data stream.

Thx.

shinjich, 2022-02-07:
Try opening the command prompt as administrator or changing the current directory.

qm13, 2022-02-10:
This error is occurs because you are writing into the tools directory under Program Files. In general it is a bad practice to write to  Program Files and Windows now prevents none administrative users from writing to any systems directory. Instead include a path with the filename to an appropriate user directory.

pythonsql21, 2022-02-14:
Thank you for your responses. It worked after I changed the directory.

closed 2022-04-18 by dasparli

haotian21, 2023-08-24:
> Thank you for your responses. It worked after I changed the directory.

Hi, can you please illustrate how I can change the directory where the recordings are saved? I am encounting the same problem right now. Thank you very much.