microsoft/Azure-Kinect-Sensor-SDK #1740: missing k4a_export.h

MadeByOzzy, 2022-02-05:
Hello!  I am trying to build this project from scratch using the supplied CMakeLists.txt file.

The file "k4a.h" references two other headers: "k4aversion.h" and "k4a_export.h".

k4aversion.h.in exists in "./include/k4a", but there is no configure_file() in the supplied CMakeLists.txt to turn the ".h.in" file in to a regular ".h" file.  Adding a configure_file() command to my upper level CMakeLists.txt file seems to have fixed that issue.

However there is still the issue of a missing k4a_export.h.  There is not a ".h.in" either, no where in the repo.

I do not know where the k4a_export.h file might be...

MadeByOzzy, 2022-02-05:
Resolved this issue.  I had to open the kinect cmake project itself, build that, that created some stuff in "./out/build/..." including the k4a_export.h and k4aversion.h files.

I had to add:

"out/build/windows-default/src/sdk/include" to my main CMakeLists.txt target_include_directories(), and that solved it

closed 2022-02-05 by MadeByOzzy

Jayden9912, 2022-04-07:
Hi @MadeByOzzy . Could you guide me on your solution? I am facing the same issue but couldn't figure out how to do it. Thanks in advance.