microsoft/Azure-Kinect-Sensor-SDK #1741: k4abt_simple_3d_viewer immediately close when open it

pramasenja, 2022-02-07:
Hi everyone.
When i open the k4abt_simple_3d_viewer.exe on Azure Kinect Body Tracking SDK, the command prompt is appear briefly, then immediately close.
Then i tried to open it on command prompt, and it showed error like this :

[image]

Then i tried to enabling logging to a file k4a.log with K4A_LOG_LEVEL=w on Environment variables and this is result :
```
New logging session started

[2022-02-07 16:36:13.055] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (225): ******************** Device Info ********************
[2022-02-07 16:36:13.055] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (226): K4A SDK version:     1.4.1
[2022-02-07 16:36:13.080] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (232): Serial Number:       00006531391
[2022-02-07 16:36:13.080] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (236): RGB Sensor Version:  1.6.110
[2022-02-07 16:36:13.080] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (239): Depth Sensor Version:1.6.79
[2022-02-07 16:36:13.080] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (242): Mic Array Version:   1.6.14
[2022-02-07 16:36:13.080] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (245): Sensor Config:       6109.7
[2022-02-07 16:36:13.080] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (248): Build type:          Release
[2022-02-07 16:36:13.080] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (253): Signature type:      MSFT
[2022-02-07 16:36:13.080] [critical] [t=5056] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (259): ****************************************************

```

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Home Build 19044.1466
 - Sensor SDK Version: 1.4.1
 - Body Tracking SDK Version: 1.1.0
 - RGB camera firmware Version: 1.6.110
 - Depth camera firmware Version: 1.6.79
 - Depth config file: 6109.7
 - Audio firmware version: 1.6.14
 - Nvidia Driver version : 511.65
 - Nvidia CUDA 11.6.99

i currently using i7 7700 and RTX 3080 Ti

**Additional context**
I tried to downgrade the Nvidia driver to 472.47, 472.12, 472.84, 471.68 the issue is still the same. 
i tried to downgrade the Body Tracking SDK to 1.0.1, and it only show this and it is stuck
[image]
already tried press k, b, or ESC, nothing happened. It's freezing.

Thanks for your help in advance!

pramasenja, 2022-02-08:
After installed CUDA Toolkit 11.6 from https://developer.nvidia.com/cuda-downloads
The k4abt_simple_3d_viewer now showing a window with title "3D Visualization" with empty/white blank. And its freezing (Not Responding).
Then after 5 seconds, k4abt_simple_3d_viewer and 3D Visualization window is closing itself.

[image]

Sagar-CK, 2022-02-21:
Same issue here!

majmas, 2022-02-24:
I have the same issue!

wleisenr, 2022-02-25:
I also have the same issue. Any workarounds or feedback on this?

fkavache, 2022-02-25:
In most cases the reason for k4abt_simple_3d_viewer crashing is the NVIDIA driver version.
NVIDIA drivers after version:

**472.12** for Game Ready drivers
and
**472.84** for Studio drivers

are having some problems with OpenGL.

@pramasenja could you please verify that after the CUDA update you are on either of these two versions?

wleisenr, 2022-02-25:
@fkavache This is the error seen in visual studio:

Unhandled exception at 0x00007FF71BF92E38 in k4abt_simple_3d_viewer.exe: An invalid parameter was passed to a function that considers invalid parameters fatal.

I have Nvidia GeForce GTX 1660 Ti, driver version 511.79.  

This issue started about two weeks ago and had been running fine since August, 2020 running SDK 1.0.1.  I had an older driver version until today when I updated it and the SDK to 1.1 hoping it would resolve the issue.  Do you recommend that I roll back Nvidia driver?

wleisenr, 2022-02-25:
@fkavache I rolled back to game ready driver 471.92 and it is now working again, thank you!

pramasenja, 2022-03-07:
Thank you @fkavache , After installing 472.84 Studio drivers, it works now.

[image]

I'm gonna close this issue

closed 2022-03-07 by pramasenja