microsoft/Azure-Kinect-Sensor-SDK #1742: Cant run install-opencv .ps1 on Windows 10 Powershell

jiany-ctrl, 2022-02-09:
Cant run install-opencv .ps1 on Windows 10 Powershell and the error message is shown below:

An exception was thrown: A parameter cannot be found that matches parameter name 'MaximumRetryCount'.
Processing downloaded file: error_Download-ToTemp_did_not_return_a_file

[image]

fkavache, 2022-02-14:
Hi @jiany-ctrl

is this related to Kinect?

jiany-ctrl, 2022-02-23:
> 

Is it in this repo?

qiaoxu123, 2022-04-02:
Hi, i meet the same problem recently and solved it successfully. It should results from the low PowerShell version.

You can type `$host` to see the current version, maybe it's lower than 7.0(like the default 5.1). The parameters like `MaximumRetryCount` added in the new version 7.0 maybe. It can be checked in this [link](https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.utility/invoke-restmethod?view=powershell-7.2&viewFallbackFrom=powershell-6)

The update command in Powershell
```
iex "& { $(irm https://aka.ms/install-powershell.ps1) } -UseMSI"
```

closed 2022-05-13 by qm13