microsoft/Azure-Kinect-Sensor-SDK #1747: Point cloud and 2d_to_3d_pixel functions

ridasaleem0, 2022-02-14:
Hi,
I am using **k4a_transformation_depth_image_to_color_camera()** to tranform my depth image to color camera geometry and then using **k4a_calibration_2d_to_3d(color, color)** to get the world coordinates of the image. 
Documentation suggest that we can alternatively use **k4a_transformation_depth_image_to_point_cloud(color)** instaed of **k4a_calibration_2d_to_3d(color, color)**, however my frames return world coordinates drastically different from one another. 

I require using point cloud instead of 2d_to_3d function in my project but my coordinate values are not matching here. If anyone could suggest points or clear this out for me. Thank you.

ridasaleem0, 2022-02-14:
@qm13

ridasaleem0, 2022-02-15:
@UnaNancyOwen any leads?

UnaNancyOwen, 2022-02-16:
@ridasaleem0 You need to attach simple code and example output for everyone can reploduce your problem.

ridasaleem0, 2022-02-16:
Thank you for your quick response. 
I have a few source pixels from the color coordinate system and want to use point cloud to generate corresponding depth values against them, just like we do using k4a_calibration_2d_to_3d(color, color). 
I am not sure if I am accessing the point cloud values correctly or not since I am pretty new to using SDK. Any help would be appreciated.

# Get a depth image but transformed in the color space.
        depth_frame_transformed = transform.depth_image_to_color_camera(depth_frame)
        
# Get Point Cloud
        xyz_image = transform.depth_image_to_point_cloud(depth_frame_transformed, k4a.ECalibrationType.COLOR)
        xyz_data = xyz_image.data.reshape(height * width, channels)
        point_cloud = xyz_data.astype(np.float64)

# loops through 17-pixel points from the color frame.
        for k in pixel_coord.keys():
                x = pixel_coord[k][0]                                              #x value in color frame
                y = pixel_coord[k][1]                                              #y value in color frame
        
                wc_x = point_cloud[x, 0]                                        # corresponding x value in point cloud
                wc_y = point_cloud[y, 1]                                        # corresponding y value in point cloud
                depth = depth_frame_transformed[x][y]                # distance value from transformed depth frame

                point = (wc_x, wc_y, depth)                                  # required (x,y,z) point in world coordinate system

UnaNancyOwen, 2022-02-16:
> I have a few source pixels from the color coordinate system and want to use point cloud to generate corresponding depth values against them

I don't think you need to via point cloud to make it happen.
The transformed depth image that transforme using `k4a_transformation_depth_image_to_color_camera()` has the same coordinate system as the color image. The depth value (z) of pixel (x, y) on color image is the value of pixel (x, y) on depth image.

ridasaleem0, 2022-02-16:
@UnaNancyOwen I require (x,y,z) in the world coordinates system and not just the distance(z-axis). 
Just as k4a_calibration_2d_to_3d(color, color) function returns, a 3D point. I am hoping to achieve the same using point cloud.

UnaNancyOwen, 2022-02-16:
You can get point cloud using `k4a_transformation_depth_image_to_point_cloud()` with `K4A_CALIBRATION_TYPE_COLOR`. That point cloud have same width and height of color image.
A point (x, y, z) corresponding to any pixel (cx, cy) on color image can be accessed buffer by the index "cy * (width * 3) + cx * 3".
(The following code has not been tested.)

```cpp
// define color pixel (e.g. center pixel of color image)
const int32_t cx = color_image.get_width_pixels() / 2;
const int32_t cy = color_image.get_height_pixels() / 2;

// calc index
const int32_t point_size = 3; // num of element of one point
const int32_t width_step = point_cloud.get_width_pixels() * point_size;
const int32_t index = cy * width_step + cx * point_size;
 
// get point
const int16_t* buffer = reinterpret_cast<int16_t*>( point_cloud.get_buffer() );
const float wx = buffer[index + 0];
const float wy = buffer[index + 1];
const float wz = buffer[index + 2];
```

ridasaleem0, 2022-02-16:
@UnaNancyOwen Thank you for your assistance.
However, I am not sure how we can replicate the buffer part for python API.

I am getting the 'index out of bound' error for the "get point" part.

UnaNancyOwen, 2022-02-17:
In the case of python, You can access a point (x, y, z) corresponding to any pixel (cx, cy) on color image by following.

```python
# get depth image aligned to color image
transformed_depth_image = transformation.depth_image_to_color_camera(depth_image)

# get point cloud
xyz_image = transformation.depth_image_to_point_cloud(transformed_depth_image, k4a.ECalibrationType.COLOR)
points = self._xyz_image.data
print(points.shape) # (720, 1280, 3)

# get color image
colors = self._color_image.data
print(colors.shape) # (720, 1280, 4)

# specify pixel (cx, cy) on color image (e.g. center of color image)
height, width, _ = colors.shape
cx = int(width / 2) # 640
cy = int(height / 2) # 360

# get point (wx, wy, wz) corresponding to specify pixel (cx, cy)
# units for point (wx, wy, wz) are [mm]
wx = points[cy, cx, 0]
wy = points[cy, cx, 1]
wz = points[cy, cx, 2]

# show pixel (cx, cy) and point (wx, wy, wz)
# if corresponding point doesn't exist, it is (0, 0, 0)
print("({}, {}) = ({}, {}, {})".format(cx, cy, wx, wy, wz)) # (640, 360) = (wx, wy, wz)
```

ridasaleem0, 2022-02-17:
@UnaNancyOwen Thank you. It works like a charm.

closed 2022-02-17 by ridasaleem0