microsoft/Azure-Kinect-Sensor-SDK #1750: capturesync_drop, releasing capture early due to full queue

hahne, 2022-02-28:
**Describe the bug**
Azure Kinect stops working after a while. Please see the log. Not sure if it is a hardware defect. The only ways to get it working again is by either restarting the PC or by power cycling the Kinect.

**To Reproduce**
It can not be directly be reproduced. I just run custom code to record videos multiple times a day and it stops working after a while. But I also saw it happening runnning k4aviewer.

**Logs**
[2022-02-28 08:55:44.192] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    200000 type:Depth
[2022-02-28 08:55:44.236] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    320266 type:Depth
[2022-02-28 08:55:44.302] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    386888 type:Depth
...
[2022-02-28 09:55:00.272] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3554986911 type:Depth
[2022-02-28 09:55:00.339] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555053600 type:Depth
[2022-02-28 09:55:00.406] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555120244 type:Depth
[2022-02-28 09:55:00.472] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555186933 type:Depth
[2022-02-28 09:55:00.539] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555253577 type:Depth
[2022-02-28 09:55:00.605] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555320266 type:Depth
[2022-02-28 09:55:00.672] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555386911 type:Depth
[2022-02-28 09:55:00.739] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555453600 type:Depth
[2022-02-28 09:55:00.806] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555520244 type:Depth
[2022-02-28 09:55:00.872] [error] [t=21475] ../src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:3555586933 type:Depth

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04.6 LTS on Nvidia Jetson Nano
 - SDK Version:  build from latest develop brnach
 - Firmware version: 
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.80
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

hahne, 2022-03-16:
Any idea what could be wrong here? Can it be a hardware defect?

aztsdfhj, 2022-03-18:
My device has also encountered such a problem recently, is there any good solution?

hahne, 2022-03-21:
There are many issues like this. It looks to me like Microsoft has more or less abandoned the whole project. Just check the commits. Last one was in October. No one seems to be working on this repo anymore.

eidetic-av, 2022-04-08:
It could be related to power. This was happening to me from a USB port that wasn't supplying enough amps. I changed to the USB port on the motherboard and was able to get color & depth images. I'm not sure that explains how this is remedied by restarting your computer though, hahne.

dasparli, 2022-04-12:
In most cases the unit will need to be powered by the round DC power cable and USB power adapter. Both of these are included in the box.
The LED on the back of the unit, between the USB and DC sockets, should be white when the proper amount of power is supplied and the USB-C cable is connected to a working USB host. If this LED is yellow then it is not getting enough power.

hahne, 2022-04-19:
I have it always powered by the round DC power cable.

vhvkhoa, 2023-01-30:
Not sure why, but this fixes the problem in my case:
Firstly, I tried turning on `k4aviewer` and open the device. Then, I set depth mode to Passive IR and start the device, the camera turns back on.
Finally, I set depth mode to any other modes and it works normally.

GuangfuWang, 2024-01-26:
in my case, i change color camera format to YUV2, then it works, for anyone that may interest.

jiapei100, 2024-07-26:
As of today, this issue persistent. Any solution found???