microsoft/Azure-Kinect-Sensor-SDK #1751: Instability when looking for bones

ekdrnqkdrn, 2022-03-01:
**Describe the bug**
The skeleton was not clearly visible when wearing the mask. 
And there is a phenomenon that the skeleton does not hang at all when lying down.
Could you please tell me how I can solve this problem?
I will be very grateful.

**To Reproduce**

ekdrnqkdrn, 2022-03-01:
[image]

ekdrnqkdrn, 2022-03-01:
The picture didn't upload, so I'll post it here.

ekdrnqkdrn, 2022-03-01:
[image]

ekdrnqkdrn, 2022-03-01:
[image]

Rudizzzz, 2022-03-01:
This is related to my problem #1634  posted 8 months ago..

owen2t, 2022-04-12:
In my experience, the body tracking works best when wearing tight fitting clothes, perhaps that would help. 

I have also noticed that the body tracking performs poorly when the person is lying down/doing a pushup. My hypothesis about why this is happening is that the skeleton is estimated using a neural network, which means it is only as good as its training data- and I don't think there are a lot of examples of a person lying down in the training data, so the body tracking will usually perform poorly. But that's just my hypothesis so don't take it as fact!

fkavache, 2022-04-14:
Hi @ekdrnqkdrn

I'm wearing a facemask too, maybe it depends on mask material, in your case BT is unable to detect depth on the mask area.

[image]

ekdrnqkdrn, 2022-05-01:
I noticed that the recognition rate dropped significantly when I wore a black mask.
Is there any way to increase the recognition rate?

closed 2022-05-01 by ekdrnqkdrn