microsoft/Azure-Kinect-Sensor-SDK #1752: C# BodySkeleton switch

Elliot8212, 2022-03-02:
I'm trying to do a single user body tracker from Kinect Azure, 
but when another person come inside camera view, my tracking jump to this person. 

Is it a bug ? How can i keep only first seen user ?

Here is the script :

```
float[] DataX = new float[32];
float[] DataY = new float[32];

var skeleton = frame.GetBodySkeleton(0);

for(int i=0;i<32;i++)
     {
       DataX[i] = skeleton.GetJoint(i).Position.X;
       DataY[i] = skeleton.GetJoint(i).Position.Y;
     }

```

Thanks

qm13, 2022-03-03:
Can you confirm that the skeleton ID for the first person in view switches to the second person when they come into view?

Another simple test is to use the k4abt_simple_3d_viewer tool and test whether the color of the first person changes when the second person comes in view.

Elliot8212, 2022-03-03:
It don’t switch in viewer, but it switch with index (0) in script

qm13, 2022-03-07:
Please compare your code with the sinple_3d_viewer source at https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/simple_3d_viewer

closed 2022-03-21 by qm13