microsoft/Azure-Kinect-Sensor-SDK #1754: Running Azure Kinect Samples. Project builds, but camera doesn't appear.

aiwhoo, 2022-03-13:
**Describe the bug**
I've setup a new machine with Windows Visual Studio 2022. Installed Azure Kinect Viewer 1.4.1 and am able to view the camera in the viewer; however, when I open up a body tracking sample (particularly, the jump_analysis_sample), I'm able to see the console display

```
 Basic Usage:

 1. Make sure you place the camera parallel to the floor and there is only one person in the scene.
 2. Raise both of your hands above your head or hit 'space' key to start the jump session.
 3. Perform a jump. Try to land at the same location as the starting point.
 4. Raise both of your hands above your head or hit 'space' key again to finish the session.
 5. Three 3d windows will pop up to show the moment of your deepest squat, jump peak and a replay of your full jump session.
    Your jump analysis results will also be printed out on the command prompt.
 6. Close any of the 3d windows to go back to the idle stage.

```

However, there is no camera. All I see is a black terminal. There are no bugs/errors/messages in the output console.

Does anyone have any suggestions as to what the issue could be?

I'm using the following desktop details:

Device name	LAPTOP-5UENRLQG
Processor	11th Gen Intel(R) Core(TM) i9-11900H @ 2.50GHz   2.50 GHz
Installed RAM	16.0 GB (15.7 GB usable)
Device ID	497A6C2D-897B-4DD6-BBD1-DE7C0D00F6A3
Product ID	00325-82188-73736-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

Edition	Windows 10 Home
Version	20H2
Installed on	‎11/‎23/‎2021
OS build	19042.1466
Experience	Windows Feature Experience Pack 120.2212.3920.0

Nuget:
Microsoft.Azure.Kinect.BodyTracking 1.0.1
Microsoft.Azure.Kinect.BodyTracking.Dependencies 0.9.1
MicrosoftAzure.Kinect.BodyTracking.Dependencies.cuDNN 0.9.1
Microsoft.Azure.Kinect.Sensor 1.3.0 (this is default, though I did try with 1.4.1)

qm13, 2022-03-14:
Please try Body Tracking v1.1.0. Note that there is NO NuGet package for this release,

zwxxx121, 2022-03-19:
@qm13 I've been using v1.1.0, but still cannot run this sample, the same problem as mentioned. Is there any other solution?

aiwhoo, 2022-03-22:
@qm13, any other suggestions to address this?

qm13, 2022-03-22:
@zwxxx121 we just released v1.1.1. Please retry with the latest SDK.

zwxxx121, 2022-03-27:
@qm13 Thanks. I've tried with the latest version, and it works!
@aiwhoo If you're using the new version, you need to update the scripts of the file. reference: https://github.com/microsoft/Azure-Kinect-Samples/pull/59/files

aiwhoo, 2022-04-01:
@zwxxx121 @qm13 Thank you. Issue has been resolved!

closed 2022-04-01 by aiwhoo