microsoft/Azure-Kinect-Sensor-SDK #1760: Displaying Pelvis Coordinate System in Python

owen2t, 2022-03-29:
I'm trying to visualize the pelvis orientation outputted by the body tracking sdk in python. I'm eventually going to be using the rest of the lower body orientations but I figured it would be best to start with joint 0. 

To calculate the x, y, and z axis of the pelvis, I transform the global kinect orientation vectors [1,0,0], [0,1,0], and [0,0,1] by the quaternion given for the pelvis. I do this using the conjugation equation p'=q * p * q^-1 where q is the quaternion and p is the vector. When I plot these rotated axes over my skeleton (first image) they don't line up with what is shown in the k4a viewer (second image). 

Is there some additional transformation that's required? Or am I misunderstanding how to apply the quaternions? I would really appreciate some guidance. 

[image]
[image]

owen2t, 2022-04-12:
It's now been 2 full weeks since I posted this question. I understand that avateering is one of the focuses of the azure kinect, but there are people and businesses who want to use the kinect for other purposes.

As mentioned in #1414 , there is insufficient documentation on how to actually use the quaternions to get coordinate systems for each body part. I was able to run the kinect in unity and all of the axes defined by the quaternions displayed correctly, so I know it's not something wrong with the body tracking- and I looked through the code for the body tracking sample and there are several transformations being done on the quaternions before they are applied. 

I am asking for some documentation, provided either in the form of an answer to this issue or an addition to the existing set of kinect samples, on how to use the quaterions outside of avateering software. Thank you.

Wilberttt, 2022-04-24:
Could you please show the Python code to visualize the quaternion?

owen2t, 2022-04-25:
Sure. First, I apply the joint's quaternion to [1,0,0], [0,1,0], and [0,0,1] with quaternion multiplication. I normalize the resulting vectors and store them in a 3x3 np array in the format [[x vector],[y_vector],[z_vector]]. That array is called xyz. The function that I have to plot takes the origin (for example, the x, y, and z coordinates of the joint) and xyz np array and plots it like this:
`axeslength = 100
x = axeslength * xyz[0,:] + origin
xvec = np.array([x,origin])
y = axeslength * xyz[1,:] + origin
yvec = np.array([y,origin])
z = axeslength * xyz[2,:] + origin
zvec = np.array([z,origin])
ax.scatter(origin[0],origin[1],origin[2],color='black')

ax.plot(xvec[:,0],xvec[:,1],xvec[:,2],color='red',label='x-axis')
ax.plot(yvec[:,0],yvec[:,1],yvec[:,2],color='green',label='y-axis')
ax.plot(zvec[:,0],zvec[:,1],zvec[:,2],color='blue',label='z-axis')`

owen2t, 2022-04-25:
The application of the quaternion is where something is going wrong and that's what I'm looking for clarity on- the plotting part works fine.

Wilberttt, 2022-04-27:
You can try axisymmetric inversion of the coordinates provided by the SDK. Compare the flipped image with the one you drew in Python, and you'll get the same result.

owen2t, 2022-04-27:
I'm not quite sure what you mean by that- Do you mean inverting the x, y, and z values of the joint location coordinates?

Wilberttt, 2022-04-28:
The image you see in body tracking viewer is mirrored I guess. And the coordinate drwan in Python is correct.

qm13, 2022-05-12:
The body tracker viewer mirrors the image.