microsoft/Azure-Kinect-Sensor-SDK #1761: Body Tracking broken on new Nvidia GPUs

gradientLord, 2022-04-02:
Just got a new 3080ti system for Kinect development, none of the body tracking samples will start.

Here are the behaviors I'm getting for each mode:

CUDA:
it spits out this error, and then closes:
2022-04-02 16:42:41.9449232 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 onnxruntime::CUDAExecutionProvider::GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
2022-04-02 16:42:41.9484857 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 onnxruntime::GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40

CPU:
-no error, it just opens and closes

DirectML:
-no error, it just opens and closes

TensorRT:
2022-04-02 16:52:07.6850310 [E:onnxruntime:K4ABT_ORT_Environment, tensorrt_execution_provider.h:45 onnxruntime::TensorrtLogger::log] [2022-04-02 21:52:07   ERROR] INVALID_ARGUMENT: getPluginCreator could not find plugin Sign version 1
Unsupported ONNX data type: DOUBLE (11)
Unsupported ONNX data type: DOUBLE (11)
Unsupported ONNX data type: DOUBLE (11)
Unsupported ONNX data type: DOUBLE (11)
2022-04-02 16:52:22.5834758 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 onnxruntime::CUDAExecutionProvider::GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign

---

I've tried upgrading and downgrading drivers (can't downgrade below Driver 511.65), because it's a new graphics card. It's important I get this up and running for a project I'm working on, are there any options to get this working on my system?

aiwhoo, 2022-04-03:
I'm noticing a similar issue on a new laptop/new install of SDKs.

An error occurred while trying to restore packages: Could not find a part of the path ...\packages\Microsoft.Azure.Kinect.BodyTracking.ONNXRuntime.1.10.0\build/netstandard2.0/Microsoft.Azure.Kinect.BodyTracking.ONNXRuntime.targets.

gradientLord, 2022-04-03:
@aiwhoo were you able to find a solution/workaround?

aiwhoo, 2022-04-03:
Unfortunately, no. Previous git issues indicate the sdk just needs to be updated.

vilmire, 2022-04-04:
제발 좀 고치자 진짜 jot같네 키넥트 c8 zz

gradientLord, 2022-04-06:
It seems people are getting the same behaviors with older cards with the latest Nvidia drivers.

fkavache, 2022-04-07:
Hi @gradientLord 
could you please take a look at this thread?

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1763#issuecomment-1091234984

aiwhoo, 2022-04-07:
@gradientLord @fkavache Kinect viewer works for me; however the sample code fails. Previous threads indicate an issue with the SDK.

fkavache, 2022-04-07:
@aiwhoo

So the simple_3d_viewer from SDK worked for you before, but the samples did not?

aiwhoo, 2022-04-10:
@fkavache That is correct.

fkavache, 2022-04-10:
@aiwhoo

It’s a different kind of issue, it seems like it was unable to find **onnxruntime.dll** in your packages and searched it in SYSTEM32 where there was an older version.

Which version of BT SDK are you using?

aiwhoo, 2022-04-10:
@fkavache the most recent--fresh download and install on a new computer.

aiwhoo, 2022-04-14:
@fkavache Any other suggestions on how to fix this issue? I've tried with older and newer onnxruntime.dll. Nuget seems to be broken with the 1.10, so I've even tried manually inserting in packages/system32

fkavache, 2022-04-14:
Hi @aiwhoo

Latest BT SDK v1.1.1 version comes with nuget package.

https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download

gradientLord, 2022-04-19:
> Hi @gradientLord could you please take a look at this thread?
> 
> [#1763 (comment)](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1763#issuecomment-1091234984)

the new body tracking viewer in here works for me, I wasn't able to get the Tensor RT cache working yet but everything else seems good!

qm13, 2022-05-25:
Fixed in v1.1.2

closed 2022-05-25 by qm13