microsoft/Azure-Kinect-Sensor-SDK #1763: Body Tracking Viewer of Azure Kinect Body Tracking SDK v1.1.1 does not launch correctly

UnaNancyOwen, 2022-04-06:
**Describe the bug**

k4abt_simple_3d_viewer.exe of Azure Kinect Body Tracking SDK v1.1.1 does not launch correctly.

**To Reproduce**

1. Install Azure Kinect Body Tracking SDK v1.1.1 by msi installer.
2. Run tools/k4abAt_simple_3d_viewer.exe.
3. Show white window with console window.
4. App exits with an error code (3221226505). 

**Expected behavior**

**Logs**

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 11 Pro x64 21H2 (22000.556) 
 - SDK Version: Azure Kinect Sensor SDK v1.4.1 and Azure Kinect Body Tracking SDK v1.1.1 
 - Firmware version: AzureKinectDK_Fw_1.6.110079014.bin (Azure Kinect Sensor SDK v1.4.1) 
 - GPU and Driver: NVIDIA GeForce GTX 1080 Ti / NVIDIA Driver 512.15 (Game Ready)

**Additional context**

fkavache, 2022-04-07:
Hi, we are working on that issue, 

Before official release could you please try if this prebuilt **k4abt_simple_3d_viewer.exe** works for you? 
just replace existing **..\Azure Kinect Body Tracking SDK\tools\k4abt_simple_3d_viewer.exe** with attached file and run body tracking with CUDA or TRT

Please use v1.1.1 BT SDK, NVidia driver version doesn't matter.

[k4abt_simple_3d_viewer.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/8440756/k4abt_simple_3d_viewer.zip)

UnaNancyOwen, 2022-04-07:
@fkavache Thanks for working on this issue! 👍 

> [k4abt_simple_3d_viewer.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/8440756/k4abt_simple_3d_viewer.zip)

It works well in my environment with DirectML and CUDA processing mode!
It doese not works with TensorRT processing mode. (I'm not sure if that was the intended behavior.)

[image]
[image]

fkavache, 2022-04-07:
@UnaNancyOwen

Glad to hear that, TRT just needs some time to build local cache.

UnaNancyOwen, 2022-04-07:
@fkavache After a few minnits wait, it works well with TensorRT processing mode. All processing modes works correctory! Thanks!

UnaNancyOwen, 2022-04-07:
However, the cache seems to be generated each time of started
[image]
.

fkavache, 2022-04-07:
It's a normal behavior, you just need to enable caching 

https://onnxruntime.ai/docs/execution-providers/TensorRT-ExecutionProvider.html#configurations

ORT_TENSORRT_CACHE_PATH=/path/to/calibration/table
ORT_TENSORRT_ENGINE_CACHE_ENABLE=1

UnaNancyOwen, 2022-04-07:
> ORT_TENSORRT_CACHE_PATH=/path/to/calibration/table
> ORT_TENSORRT_ENGINE_CACHE_ENABLE=1

Oh! It is good! 👍

leventalpsal, 2022-04-07:
> Hi, we are working on that issue,
> 
> Before official release could you please try if this prebuilt **k4abt_simple_3d_viewer.exe** works for you? just replace existing **..\Azure Kinect Body Tracking SDK\tools\k4abt_simple_3d_viewer.exe** with attached file and run body tracking with CUDA or TRT
> 
> Please use v1.1.1 BT SDK and the latest NVidia driver.
> 
> [k4abt_simple_3d_viewer.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/8440756/k4abt_simple_3d_viewer.zip)

I had the same issue, Azure Kinect Viewer working, k4abt_simple_3d_viewer.exe (2133KB) crashing on startup. The prebuilt k4abt_simple_3d_viewer.exe (2134KB) works fine.

Environment details:
Azure Kinect SDK 1.4.1
Azure Kinect Body Tracking SDK 1.1.1
Ryzen 9 5900 HX
Nvidia RTX 3070 laptop GPU
Nvidia Game Ready driver 511.79
Windows 11 22000.593

gradientLord, 2022-04-07:
> It's a normal behavior, you just need to enable caching
> 
> https://onnxruntime.ai/docs/execution-providers/TensorRT-ExecutionProvider.html#configurations
> 
> ORT_TENSORRT_CACHE_PATH=/path/to/calibration/table ORT_TENSORRT_ENGINE_CACHE_ENABLE=1

how exactly do you enable TENSORRT caching on your system? I tried these commands in elevated command prompt and it didnt work.

fkavache, 2022-04-14:
Hi, @gradientLord

Those are environment variables, try:
**$env:ORT_TENSORRT_ENGINE_CACHE_ENABLE = 1
$env:ORT_TENSORRT_CACHE_PATH="C:\Users\\...\Desktop\cache"** from powershell.

UnaNancyOwen, 2022-04-14:
> TensorrtExecutionProvider_TRTKernel_graph_tf2onnx_10554548234644750994_1_0.engine

If set up correctly, it will generate a cache file like this one in first run time! And the next time, the viewer application will be launched quickly. 👍

Aquariusquintini, 2022-04-22:
> Hi, we are working on that issue,
>
> Before official release could you please try if this prebuilt **k4abt_simple_3d_viewer.exe** works for you? just replace existing **..\Azure Kinect Body Tracking [...]

I can confirm that the attached zip-file works.
Environment details:
Azure Kinect SDK 1.4.1
Azure Kinect Body Tracking SDK 1.1.1
Intel Core i9-10980XE
Nvidia RTX 3080 GPU
Nvidia driver 512.15
Windows 10 21H1 (Build 19043.1526)

pschaub, 2022-05-06:
> Hi, we are working on that issue,
>
> Before official release could you please try if this prebuilt **k4abt_simple_3d_viewer.exe** works for you? just replace existing **..\Azure Kinect Body Tracking [...]

works, tested on RTX 3090, nvidia driver 512.15

qm13, 2022-05-25:
Fixed in v1.1.2

closed 2022-05-25 by qm13

anggawahyuw, 2022-11-28:
> 
It doesn't work for my laptop, RTX 3070 TI. I am confused how to run body tracking with CUDA or TRT?

anggawahyuw, 2022-11-28:
it works, RTX 3070 Ti, windows 11 pro, thank you