microsoft/Azure-Kinect-Sensor-SDK #1764: Failure in compiling examples/calibration/main.cpp

Jayden9912, 2022-04-07:
Hi. Thanks in advance for spending time reading my issue.

Could anyone help me on this error?

The OS system using is Ubuntu 16, and the SDK is built from source. 
I rename the main.cpp to camera_calibration.cpp and put it to the root of Azure-Kinect-Sensor-SDK. The command use is
cd to the root of Azure-Kinect-Sensor-SDK
```
g++ -Iinclude -std=c++11 camera_calibration.cpp
```

Here is the error:
[image]

The file structure 
Azure-Kinect-Sensor-SDK/include/k4a/k4a.h
Azure-Kinect-Sensor-SDK/include/k4a/k4a.hpp
Azure-Kinect-Sensor-SDK/include/k4a/k4a_export.h
Azure-Kinect-Sensor-SDK/include/k4a/k4atypes.h
Azure-Kinect-Sensor-SDK/include/k4a/k4aversion.h
Azure-Kinect-Sensor-SDK/include/k4a/k4aversion.h.in

The following is how I build this SDK.
```
git clone https://github.com/microsoft/Azure-Kinect-Sensor-SDK.git
cd Azure-Kinect-Sensor-SDK
mkdir build && cd build
cmake .. -GNinja
Ninja
```

Jayden9912, 2022-04-11:
Hi, any update on this?

fkavache, 2022-04-14:
Hi, @Jayden9912

What you are trying to do? could you please share your camera_calibration.cpp?

Jayden9912, 2022-04-14:
Hi, @fkavache . 

I am trying to compile this cpp file only. My camera_calibration.cpp is the same as the one in this repo.
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/calibration

fkavache, 2022-04-14:
@Jayden9912

You need to link k4a, this line, from cmake, is responsible for linking k4a to that example.

[image]

Jayden9912, 2022-04-14:
I am using Ubuntu 16 and I only build from source the whole repo but I didn't install it into system directory. If I use cmake, it fails. Is there any way to link the library to g++? I am more proficient in Python so I may need guidance in compiling this code. Thanks in advance for any guidance provided.

fkavache, 2022-04-14:
@Jayden9912

Could you please screen what the cmake tells you when you try this?

```
git clone https://github.com/microsoft/Azure-Kinect-Sensor-SDK.git
cd Azure-Kinect-Sensor-SDK
mkdir build && cd build
cmake .. -GNinja
Ninja
```

nndorenji, 2022-10-08:
I also face the same problem. Is there any solution for this? As @fkavache said, this is what I've got from run ninja 
[image]
and this is what's inside my cmakelist.txt 
[image]
but still getting the result as this. I think the link directory is already right. Am I missing something? 
[image]