microsoft/Azure-Kinect-Sensor-SDK #1767: Failure execute sample_unity_bodytracking Build with BT1.1.1

naonaoue, 2022-04-16:
I'm trying BT1.1.1 with Unity.
'sample_unity_bodytracking' works perfectly with UnityEditor2020.3 (LTS) if you follow these documents.

'Change log v1.1.1','Microsoft Visual C ++ Redistributable latest supported downloads', and'Azure-Kinect-Samples / body-tracking-samples / sample_unity_bodytracking / ReadMe.'

However, in the build of'PC, Mac & Linux Standalone-> windows-> x86_64', even if you insert Dll as instructed after the build is successful, when you execute it,'K4A_WAIT_RESULT_FAILED' wait the thread of BT.

Finally if you Build a Standalone Executable
#### Then you will need to put these files in the same directory with the .exe:
onnxruntime.dll
onnxruntime_providers_cuda.dll
onnxruntime_providers_shared.dll
onnxruntime_providers_tensorrt.dll
dnn_model_2_0_op11.onnx
dnn_model_2_0_lite_op11.onnx
cudnn64_8.dll
cudnn_cnn_infer64_8.dll
cudnn_ops_infer64_8.dll
cudart64_110.dll
cublas64_11.dll
cublasLt64_11.dll
cufft64_10.dll
directml.dll

why is that?

fkavache, 2022-04-17:
Hi @naonaoue

What is the question exactly? Why does a standalone executable need library files to be copied by yourself?

naonaoue, 2022-04-17:
Hi@[fkavache](https://github.com/fkavache)
Excuse me. It seems that I didn't have enough vocabulary.
I know I need a copy as it is written in the ReadMe.
If you follow the ReaMe instructions in'sample_unity_bodytracking'to copy the dll, the exe built by Unity will not work.

fkavache, 2022-04-17:
@naonaoue 

Could you please share how to reproduce exactly?

naonaoue, 2022-04-17:
Premise)
1)My Local is reactor2019.
[image]
2)This Sample is Works fine On 'Unity Editor 2020.3.33f1'.
[image]
3)Build
[image]
4)copy dll files in the same directory with the .exe
[image]
5)not work.
[image]

fkavache, 2022-04-17:
@naonaoue

It works in my case, from where are you copying the library files?

[image]

naonaoue, 2022-04-17:
[fkavache](https://github.com/fkavache)
Oh, really?
Very thanks for your try!

I copying from ,
1st)C:\Program Files\Azure Kinect Body Tracking SDK\tools
2nd)body-tracking-samples\csharp_3d_viewer\bin\x64\Debug\net48\win-x64
In my case, neither was good.

I want you to tell me the correct place.

fkavache, 2022-04-17:
@naonaoue

First run **MoveLibraryFiles.bat** it will do the job for you and copy the library files from 
**...\Azure-Kinect-Samples\body-tracking-samples\sample_unity_bodytracking**

naonaoue, 2022-04-17:
@fkavache
Yes, I did that too and it works in Unity Editor.
However, I think that there are not enough files.

if not exist Assets\Plugins mkdir Assets\Plugins
set BODY_TRACKING_SDK_PATH=C:\Program Files\Azure Kinect Body Tracking SDK\
set BODY_TRACKING_TOOLS_PATH="%BODY_TRACKING_SDK_PATH%tools\"
set BODY_TRACKING_LIB_PATH="%BODY_TRACKING_SDK_PATH%sdk\netstandard2.0\release\"
copy %BODY_TRACKING_LIB_PATH%Microsoft.Azure.Kinect.BodyTracking.dll Assets\Plugins
copy %BODY_TRACKING_LIB_PATH%Microsoft.Azure.Kinect.BodyTracking.pdb Assets\Plugins
copy %BODY_TRACKING_LIB_PATH%Microsoft.Azure.Kinect.BodyTracking.deps.json Assets\Plugins
copy %BODY_TRACKING_LIB_PATH%Microsoft.Azure.Kinect.BodyTracking.xml Assets\Plugins
copy packages\Microsoft.Azure.Kinect.Sensor.1.4.1\lib\netstandard2.0\Microsoft.Azure.Kinect.Sensor.dll Assets\Plugins
copy packages\Microsoft.Azure.Kinect.Sensor.1.4.1\lib\netstandard2.0\Microsoft.Azure.Kinect.Sensor.pdb Assets\Plugins
copy packages\Microsoft.Azure.Kinect.Sensor.1.4.1\lib\netstandard2.0\Microsoft.Azure.Kinect.Sensor.deps.json Assets\Plugins
copy packages\Microsoft.Azure.Kinect.Sensor.1.4.1\lib\netstandard2.0\Microsoft.Azure.Kinect.Sensor.xml Assets\Plugins
copy %BODY_TRACKING_TOOLS_PATH%cublas64_11.dll Assets\Plugins
copy %BODY_TRACKING_TOOLS_PATH%cublasLt64_11.dll Assets\Plugins
copy %BODY_TRACKING_TOOLS_PATH%cudart64_110.dll Assets\Plugins
copy packages\System.Buffers.4.4.0\lib\netstandard2.0\System.Buffers.dll Assets\Plugins
copy packages\System.Memory.4.5.3\lib\netstandard2.0\System.Memory.dll Assets\Plugins
copy packages\System.Runtime.CompilerServices.Unsafe.4.5.2\lib\netstandard2.0\System.Runtime.CompilerServices.Unsafe.dll Assets\Plugins
copy packages\System.Reflection.Emit.Lightweight.4.6.0\lib\netstandard2.0\System.Reflection.Emit.Lightweight.dll Assets\Plugins
copy packages\Microsoft.Azure.Kinect.Sensor.1.4.1\lib\native\amd64\release\depthengine_2_0.dll Assets\Plugins
copy packages\Microsoft.Azure.Kinect.Sensor.1.4.1\lib\native\amd64\release\k4a.dll Assets\Plugins
copy packages\Microsoft.Azure.Kinect.Sensor.1.4.1\lib\native\amd64\release\k4arecord.dll Assets\Plugins
copy %BODY_TRACKING_TOOLS_PATH%onnxruntime.dll Assets\Plugins
copy %BODY_TRACKING_TOOLS_PATH%k4abt.dll Assets\Plugins
copy %BODY_TRACKING_TOOLS_PATH%cudnn64_8.dll .\
copy %BODY_TRACKING_TOOLS_PATH%cudnn_cnn_infer64_8.dll .\
copy %BODY_TRACKING_TOOLS_PATH%cudnn_ops_infer64_8.dll .\
copy %BODY_TRACKING_TOOLS_PATH%onnxruntime.dll .\
copy %BODY_TRACKING_TOOLS_PATH%dnn_model_2_0_op11.onnx .\
copy %BODY_TRACKING_TOOLS_PATH%cublas64_11.dll .\
copy %BODY_TRACKING_TOOLS_PATH%cublasLt64_11.dll .\
copy %BODY_TRACKING_TOOLS_PATH%cudart64_110.dll .\
copy %BODY_TRACKING_TOOLS_PATH%cufft64_10.dll .\

fkavache, 2022-04-17:
@naonaoue

What's happening when you open the UnityCSharp.exe?

fkavache, 2022-04-17:
@naonaoue

Try File -> Build Settings -> Build.

In my case there is no scene selected.

[image]

naonaoue, 2022-04-17:
@fkavache 
I tried "No scene selected" but unfortunately it was the same.

Using your screenshot as a reference, I put in 9 files from the root of 'sample_unity_bodytracking' and 'Assets/Plugins'.
But it didn't work.
It feels like the BT is out

https://user-images.githubusercontent.com/27718076/163721774-dd451f6f-6171-45a7-9589-5d2d8da14c0d.mp4

.

naonaoue, 2022-05-12:
I may have found another bug.
Does "k4abt.dll" recognize the path of double-byte characters?
BT did not start when I used Japanese double-byte characters in the path of the Unity executable file. "K4a.dll" seems to be compatible, and Kinect ID etc. were picked up.

qm13, 2022-05-17:
@naonaoue that is a know issue. See #1540.

naonaoue, 2022-05-18:
@qm13 Oh really! Will it be fixed in BT1.1.2?

qm13, 2022-05-25:
There are currently no plans to fix it.

closed 2022-05-25 by qm13

KikeVilla, 2022-06-01:
Somebody found some fix for this issue?. I got the same problem.