microsoft/Azure-Kinect-Sensor-SDK #1779: Azure Kinect does not connect after updating to windows 11 from windows 10.

yz-cnsdqz, 2022-05-15:
**Describe the bug**
The Azure kinect worked well on windows 10. After updating to win11, the device cannot be connected.
- The **while** LED flicks on the back of the Azure kinect. 
- Updating the firmware does not work because the device is not connected. 
- The device is already connected to USB3.0 without any hub. 
- The AzureViewer (run as admin) cannot detect the device. 

**Desktop (please complete the following information):**
 - Windows 11
 - SDK Version:  1.4.1
 - Firmware version:  (should be in the logs, or use `AzureKinectFirmwareTool.exe -q`) The firmware tool does not work. It gives
 `[2022-05-15 23:15:00.564] [error] [t=4664] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (370): find_libusb_device(). Unable to open LIBUSB at index 0
[2022-05-15 23:15:00.565] [error] [t=4664] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth_mcu\depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2022-05-15 23:15:00.588] [error] [t=4664] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (370): find_libusb_device(). Unable to open LIBUSB at index 0
[2022-05-15 23:15:00.589] [error] [t=4664] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color_mcu\color_mcu.c (74): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, device_index, NULL, &colormcu->usb_cmd) returned failure in colormcu_create_by_index()
ERROR: Unable to find a device`

vaibhavsatpathy, 2022-06-24:
Was anyone able to resolve this issue. Not able to run it on Windows 11. It keeps saying that Invalid configuration Descriptor in the Device Manager.

blackdown, 2022-10-12:
Same issue here