microsoft/Azure-Kinect-Sensor-SDK #1783: BT v1.1.2 crashes on app exit in `onnxruntime_providers_cuda.dll` or `onnxruntime_providers_tensorrt.dll`

diablodale, 2022-05-25:
Either BT v1.1.1 or v1.1.2 introduced a crash on app exit that occurs in `onnxruntime_providers_cuda.dll` and `onnxruntime_providers_tensorrt.dll`. This crash does not occur with BT v1.1.0.

If an app only uses DirectML. No crash on app exit.
If an app only uses CUDA. No crash on app exit.
If an app only uses TensorRT. No crash on app exit.
But...
When an app uses DirectML, shutdown+destroy k4a/tracker, uses CUDA --> crash in `onnxruntime_providers_cuda.dll`
When an app uses DirectML, shutdown+destroy k4a/tracker, uses TensorRT --> crash in `onnxruntime_providers_tensorrt.dll`

### Setup

* Microsoft Windows [Version 10.0.19044.1706]
* VS2019 community v16.11.14
* kinect sdk v1.4.1
* kinect bt sdk v1.1.2
* laptop intel i7-10875h
* nvidia rtx2070 super (directml device 0)
* intel uhd graphics (directml device 1)
*  == Azure Kinect DK Firmware Tool ==
  Device Serial Number: 000035692012
  Current Firmware Versions:
  RGB camera firmware:      1.6.110
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

### Repro

1. setup as above
2. code an app that displays kinect rgb video and uses bt to track the right wrist. The app should have the ability to start/stop the video+bt and run it on any cpu,directml,cuda,tensorrt,tensorrt16 *without* stopping the app
3. config and build for debug, x64, windows
4. Run app.
5. display rgb video and track the right wrist using DirectML
6. stop displaying and tracking. DONT STOP THE APP.
7. change to CUDA for bt
8. display rgb video and track the right wrist using CUDA
9. exit the app

### Result

Crash exception on app exit

```
Exception thrown at 0x00007FFAEF213C1E (onnxruntime_providers_cuda.dll) in Max.exe: 0xC0000005: Access violation writing location 0x0000000052FAA758.

onnxruntime_providers_cuda.dll!00007ffaef213c1e() (Unknown Source:0)
onnxruntime.dll!00007ffb2f37c6d4() (Unknown Source:0)
onnxruntime.dll!00007ffb2f37caa4() (Unknown Source:0)
onnxruntime.dll!00007ffb2f3ab456() (Unknown Source:0)
onnxruntime.dll!00007ffb2f35c389() (Unknown Source:0)
k4abt.dll!00007ffb3aa1c079() (Unknown Source:0)
k4abt.dll!00007ffb3a9ada4f() (Unknown Source:0)
k4abt.dll!00007ffb3a9ad5d9() (Unknown Source:0)
k4abt.dll!00007ffb3a9adbad() (Unknown Source:0)
k4abt.dll!00007ffb3a9ab60b() (Unknown Source:0)
k4abt.dll!00007ffb3a9ab495() (Unknown Source:0)
k4abt.dll!00007ffb3a9ab79b() (Unknown Source:0)
k4abt.dll!00007ffb3a920d63() (Unknown Source:0)
k4abt.dll!00007ffb3a9218c5() (Unknown Source:0)
k4abt.dll!00007ffb3a921a0e() (Unknown Source:0)
ntdll.dll!LdrpCallInitRoutine() (Unknown Source:0)
ntdll.dll!LdrShutdownProcess() (Unknown Source:0)
ntdll.dll!RtlExitUserProcess() (Unknown Source:0)
kernel32.dll!ExitProcessImplementation() (Unknown Source:0)
[Inline Frame] msvcr120.dll!__crtExitProcess(int) Line 774 (f:\dd\vctools\crt\crtw32\startup\crt0dat.c:774)
```

If you change step 7 and 8 to be TensorRT, then it also crashes with

```
Exception thrown at 0x00007FFB3659F71E (onnxruntime_providers_tensorrt.dll) in Max.exe: 0xC0000005: Access violation writing location 0x0000000070578268

onnxruntime_providers_tensorrt.dll!00007ffb3659f71e() (Unknown Source:0)
onnxruntime.dll!00007ffb2f37c6d4() (Unknown Source:0)
onnxruntime.dll!00007ffb2f37ca98() (Unknown Source:0)
onnxruntime.dll!00007ffb2f3ab456() (Unknown Source:0)
onnxruntime.dll!00007ffb2f35c389() (Unknown Source:0)
k4abt.dll!00007ffb36b9c079() (Unknown Source:0)
k4abt.dll!00007ffb36b2da4f() (Unknown Source:0)
k4abt.dll!00007ffb36b2d5d9() (Unknown Source:0)
k4abt.dll!00007ffb36b2dbad() (Unknown Source:0)
k4abt.dll!00007ffb36b2b60b() (Unknown Source:0)
k4abt.dll!00007ffb36b2b495() (Unknown Source:0)
k4abt.dll!00007ffb36b2b79b() (Unknown Source:0)
k4abt.dll!00007ffb36aa0d63() (Unknown Source:0)
k4abt.dll!00007ffb36aa18c5() (Unknown Source:0)
k4abt.dll!00007ffb36aa1a0e() (Unknown Source:0)
ntdll.dll!LdrpCallInitRoutine() (Unknown Source:0)
ntdll.dll!LdrShutdownProcess() (Unknown Source:0)
ntdll.dll!RtlExitUserProcess() (Unknown Source:0)
kernel32.dll!ExitProcessImplementation() (Unknown Source:0)
[Inline Frame] msvcr120.dll!__crtExitProcess(int) Line 774 (f:\dd\vctools\crt\crtw32\startup\crt0dat.c:774)
```

### Logs

Attached are two log files; for the core sdk + bt sdk. I followed the repo steps above for CUDA.

When the exception occurs, the two log files (k4a, k4abt) are both 0 byte files. I wait. Still zero byte. So I kill the app in the debugger and then the files have content.
[logs.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/8774457/logs.zip)

diablodale, 2022-05-25:
Additional cases that DO NOT crash on exit

* only CUDA
* only Tensor
* CUDA then DirectML
* TensorRT then DirectML

al-baum, 2022-09-21:
Any news on this issue ?