microsoft/Azure-Kinect-Sensor-SDK #1786: Somebody found some fix for this issue?. I have the same problem.

KikeVilla, 2022-06-02:
Somebody found some fix for this issue?. I got the same problem.

_Originally posted by @KikeVilla in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1147#issuecomment-1143131937_

KikeVilla, 2022-06-02:
AzureKinectBodyTrackingCreateException: result = K4A_RESULT_FAILED
Microsoft.Azure.Kinect.BodyTracking.Tracker.Create (Microsoft.Azure.Kinect.Sensor.Calibration calibration, Microsoft.Azure.Kinect.BodyTracking.TrackerConfiguration configuration) (at <59f09534e4254e89b28779441231cce7>:0)

fkavache, 2022-06-03:
Hi @KikeVilla 

Before official merge you can try if changes made in this [branch](https://github.com/microsoft/Azure-Kinect-Samples/tree/v-vkatsadze/unity-sample/body-tracking-samples/sample_unity_bodytracking) works for you.
Please follow the README.md there are some changes made there.

fkavache, 2022-06-03:
@KikeVilla

Merged https://github.com/microsoft/Azure-Kinect-Samples/pull/70

closed 2022-06-09 by qm13

KikeVilla, 2022-06-09:
Thanks for your help, it works now.