microsoft/Azure-Kinect-Sensor-SDK #1787: Tthe power mode of IMU inside Azure Kinect

xiaotaw, 2022-06-04:
Hello, 

[The embedded Inertial Measurement Unit (IMU) is an LSM6DSMUS](https://github.com/MicrosoftDocs/azure-docs/blob/main/articles/kinect-dk/hardware-specification.md#motion-sensor-imu). And [the datasheet of LSM6DSM]( https://www.st.com/resource/en/datasheet/lsm6dsm.pdf) says that it supports two pow consumption modes: the normal/low-power mode and high-performance mode. And several important parameters (e.g. **noise density**) differ when the device is under low or high power mode.

Here comes my problem,  I need to know which power consumption mode of the IMU is chose inside Azure Kinect?  Because in my application, the exact value of those parameters, especially **noise density**, really matters.

Thank you!

dasparli, 2022-06-09:
The IMU defaults to high-power mode and the firmware in the AKDK does not set the power mode during operation so it must be in high-power mode.

xiaotaw, 2022-06-10:
Got it, thank you!

closed 2022-06-10 by xiaotaw