microsoft/Azure-Kinect-Sensor-SDK #1789: Extrinsic parameters

pythonsql21, 2022-06-06:
Can you give me the average range of Azure Kinect extrinsic parameters i.e. rotation and translation?
I have extracted the intrinsic parameters but am not able to extract the extrinsic ones.
Thanks.

dasparli, 2022-06-06:
Here's a couple of articles that will show you how to extract the extrinsic parameters from your camera:

https://docs.microsoft.com/en-us/azure/kinect-dk/use-calibration-functions
https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga4e43940d8d8db48da266c7a7842c8d78.html#ga4e43940d8d8db48da266c7a7842c8d78

pythonsql21, 2022-06-08:
Thank you @dasparli. Could you please give me the average values of the Azure Kinect extrinsic parameters i.e. rotation and translation? Someone has asked for the intrinsic parameters and got the answer here: https://docs.microsoft.com/en-us/answers/questions/201906/pixel-size-of-rgb-and-tof-camera.html
I want the same answer as the above for the extrinsic parameters.