microsoft/Azure-Kinect-Sensor-SDK #1790: OpenSSL3.0 and Ubuntu 22.04 Support...

jiapei100, 2022-06-14:
Looking forward to the support of
OpenSSL3 and Ubuntu 22.04.

Thanks...

YellowOrz, 2022-07-12:
For ubuntu 22.04, you can download [k4a-tools_1.4.1_amd64.deb](https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/k/k4a-tools/k4a-tools_1.4.1_amd64.deb), [libk4a1.4-dev_1.4.1_amd64.deb](https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.4-dev/libk4a1.4-dev_1.4.1_amd64.deb) and [libk4a1.4_1.4.1_amd64.deb](https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.4/libk4a1.4_1.4.1_amd64.deb) for 18.04. Before installing this three .deb file, you need to temporarily switch the source list of apt to 20.04 and then install libsoundio1 (sudo apt install libsoundio1, which is  not avaliable in the source list  of 22.04.
This method works for me.

jackjansen, 2023-05-02:
> Before installing this three .deb file, you need to temporarily switch the source list of apt to 20.04 and then install libsoundio1 (sudo apt install libsoundio1, which is not available in the source list of 22.04. 

Does anyone happen to have a recipe/example to do this, especially on the github actions ci/cd?

jackjansen, 2023-05-02:
Answering my own request:

> Does anyone happen to have a recipe/example to do this, especially on the github actions ci/cd?

Here is a code snippet I use in my github actions to install k4a and k4abt on Ubuntu 22.04. It also has the code to pre-accept the EULAs.

```
sudo apt-add-repository -y -n 'deb http://archive.ubuntu.com/ubuntu focal main'
sudo apt-add-repository -y 'deb http://archive.ubuntu.com/ubuntu focal universe'
sudo apt-get install -y libsoundio1
sudo apt-add-repository -r -y -n 'deb http://archive.ubuntu.com/ubuntu focal universe'
sudo apt-add-repository -r -y 'deb http://archive.ubuntu.com/ubuntu focal main'

if ! dpkg -s libk4a1.3 > /dev/null; then
	curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.3/libk4a1.3_1.3.0_amd64.deb > /tmp/libk4a1.3_1.3.0_amd64.deb
	echo 'libk4a1.3 libk4a1.3/accepted-eula-hash string 0f5d5c5de396e4fee4c0753a21fee0c1ed726cf0316204edda484f08cb266d76' | sudo debconf-set-selections
	sudo dpkg -i /tmp/libk4a1.3_1.3.0_amd64.deb
fi
if ! dpkg -s libk4a1.3-dev > /dev/null; then
	curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.3-dev/libk4a1.3-dev_1.3.0_amd64.deb > /tmp/libk4a1.3-dev_1.3.0_amd64.deb
	sudo dpkg -i /tmp/libk4a1.3-dev_1.3.0_amd64.deb
fi
if ! dpkg -s libk4abt1.0 > /dev/null; then
	curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4abt1.0/libk4abt1.0_1.0.0_amd64.deb > /tmp/libk4abt1.0_1.0.0_amd64.deb
	echo 'libk4abt1.0	libk4abt1.0/accepted-eula-hash	string	03a13b63730639eeb6626d24fd45cf25131ee8e8e0df3f1b63f552269b176e38' | sudo debconf-set-selections
	sudo dpkg -i /tmp/libk4abt1.0_1.0.0_amd64.deb
fi
if ! dpkg -s libk4abt1.0-dev > /dev/null; then
	curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4abt1.0-dev/libk4abt1.0-dev_1.0.0_amd64.deb > /tmp/libk4abt1.0-dev_1.0.0_amd64.deb
	sudo dpkg -i /tmp/libk4abt1.0-dev_1.0.0_amd64.deb
fi
if ! dpkg -s k4a-tools > /dev/null; then
	curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/k/k4a-tools/k4a-tools_1.3.0_amd64.deb > /tmp/k4a-tools_1.3.0_amd64.deb
	sudo dpkg -i /tmp/k4a-tools_1.3.0_amd64.deb
fi
```

remmel, 2023-05-17:
@jackjansen is there a reason you are not using last version (k4a-tools1.4 libk4a1.4-dev libk4abt1.1-dev) ?

jackjansen, 2023-07-11:
Sorry @remmel I hadn't seen your message until just now. There was a historic reason for sticking with 1.3 for our application, but I must admit that I don't remember anymore what the problem was with 1.4. I vaguely remember that it had to do with the fact that k4abt 1.1 didn't work for us, and therefore we reverted to k4abt 1.0, and that required us to revert to k4a 1.3.

I will try to find the time to investigate, but I can't promise anything.