microsoft/Azure-Kinect-Sensor-SDK #1791: Is it possiable to stream the azure kinect camera and run body tracking SDK on another computer?

Cecropis, 2022-06-22:
**Is your feature request related to a problem? Please describe.**

I'm trying to steam Azure Kinect over ethernet, but It seems the body tracking SDK would be only working on the PC that the camera is directly plugged into? 
**Describe the solution you'd like**

Is there any way to make the body tracking SDK runs on another computer? If we can stream the camera via PoE and do all the calculations on the server, it will be easier to manage. 
**Describe alternatives you've considered**

**Additional context**

qm13, 2022-06-27:
An application built on the Body Tracking SDK can run on a separate PC to the PC running the Azure Kinect camera. You would need to develop code to stream the IR and depth data from the Azure Kinect PC to the BT PC. There are a couple of GitHub projects on compressing Kinect data streams as you will likely need to compress them to fit in the LAN bandwidth. See https://github.com/catid/Zdepth and https://github.com/catid/ZdepthLossy.

wei666666, 2022-07-03:
> 

> **Is your feature request related to a problem? Please describe.**
>
> I'm trying to steam Azure Kinect over ethernet, but It seems the body tracking SDK would be only working on the PC that the camera [...]

Hi, [Cecropis](https://github.com/Cecropis). Recently, I also want to utilize the RTSP-Streamer/Server to stream the Kinects RGB- and Depth-Video to network. Do you have any ideas or methods about how to achieve that.