microsoft/Azure-Kinect-Sensor-SDK #1800: Is it possible to convert 1024x1024 depth image into a 1024x1024x3 point cloud?

JeffR1992, 2022-07-12:
I'm running a series of experiments with the Azure Kinect at a depth resolution of 1024x1024, and am trying to figure out if it's possible to convert a 1024x1024 depth image into an 1024x1024x3 point cloud. That is, I'd like to convert each pixel in the 1024x1024 depth image from an depth/range into an XYZ position.

Once I get a 1024x1024x3 array of XYZ points, I'm looking to then only choose the z-coordinate and thus convert the 1024x1024x3 point cloud back into a 1024x1024 "z-depth" image instead of a vanilla depth image.

Any help would be appreciated. Thanks!

conatsera, 2022-07-12:
Each pixel in the depth image is the distance in mm on the z axis. When
creating a point cloud image, only the X and Y axes are derived by
multiplying against the calibration table for each pixel. The Z value is
copied.

The example should help for understanding the depth data and creating a
depth point cloud.
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/fastpointcloud/main.cpp#L65

JeffR1992, 2022-07-13:
Interesting, that makes my life easier. However, just so I understand correctly, if our coordinate axes are defined as follows (with y going into the page):

z
^
 |
 ----> x

are you saying that by default the depth image reports its values as follows:

^     ^    ^
 |      |      |
 |      |      |
 |      |      |

That is, completely ignoring the x and y coordinate values and only using the z coordinate value for each pixel. Instead of reporting the depth using the range along each "pixel ray" as follows (i.e. r = sqrt(x^2 + y^2 + z^2)):

^    ^    ^
  \    |    / 
    \  |  /
      \|/

conatsera, 2022-07-13:
> These measurements are processed to generate a depth map. A depth map is a set of Z-coordinate values for every pixel of the image, measured in units of millimeters.
https://docs.microsoft.com/en-us/azure/kinect-dk/depth-camera#operating-principles

The depth image doesn't contain distance from the sensor, it specifically provides z-coordinates.