microsoft/Azure-Kinect-Sensor-SDK #1801: Is the documentation related to external synchronization misleading?

mdabdk, 2022-07-12:
I am trying to view the RGB, depth, and IR outputs of two different kinects on the same Azure Kinect Viewer instance. I'm following the instructions here: [Verify that the devices are connected and communicating](https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync#verify-that-the-devices-are-connected-and-communicating). It seems that, on step 2, I should be able to view the serial numbers for both kinects, as shown below:

[image]

However, I'm not able to do so. Additionally, according to [this answer](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1527#issuecomment-797846480), it seems that this is not actually possible. Therefore, I am wondering if the documentation is misleading, or the answer is misleading.

My current setup is as follows: I have two Kinects, Kinect A and Kinect B. Kinect A has the following device firmware versions:

RGB camera: 1.6.102
Depth camera: 1.6.75
Audio: 1.6.14
Build config: Release
Signature type: Microsoft

Kinect B has the following device firmware versions:

RGB camera: 1.6.102
Depth camera: 1.6.75
Audio: 1.6.14
Build config: Release
Signature type: Microsoft

Kinect A is connected to a machine running Windows, with SDK version 1.3.0, while Kinect B is connected to a machine running Ubuntu 18.04, with SDK version 1.3.0. Both Kinect A and Kinect B have their indicator lights on at steady white, meaning that both power and data cables should be connected properly. I'm able to open the Azure Kinect Viewer on both the Windows machine and the Linux machine, and individually start the kinects on both machines. Also, Kinect A was chosen to be the master, while Kinect B was chosen to be the subordinate. Therefore, the Sync Out port of Kinect A is connected to the Sync In port of Kinect B.

To test external synchronization, I started Kinect B using the Azure Kinect Viewer as a subordinate. Then, I started Kinect A using the viewer as a Master. As soon as I start Kinect A as master, the RGB, depth, and IR streams for Kinect B open, which is expected. However, on the Windows machine that is connected to Kinect A, the master, I am not able to see the serial number for Kinect B. It seems from the documentation mention above that this should be possible, but I'm not sure.

rajkundu, 2022-07-14:
If I understand your question correctly, and if I understand how the devices/SDK work correctly, then unfortunately, no – your intended behavior is not possible.

As far as I understand, just like frame data, the serial numbers are transmitted via USB from each camera to its host computer. The connection between the cameras, on the other hand, is a 3.5 mm sync trigger. Besides this trigger, this line carries no data – neither serial numbers nor frames. So, cameras cannot see each other's serial numbers/frames, and neither can their respective host computer.

In my own use case, I have two Azure Kinect DK cameras plugged into the *same* host computer, and I do see both serial numbers in the Azure Kinect Viewer app regardless of their external sync connections. However, this makes sense, as they each are connected to this single host computer via their own individual USB connection. I'm guessing that this same 2-camera/1-host setup was used in the screenshot you cited.

mdabdk, 2022-07-14:
> If I understand your question correctly, and if I understand how the devices/SDK work correctly, then unfortunately, no – your intended behavior is not possible.
>
> As far as I understand, just like [...]

In that case, I think the documentation is indeed misleading, because the preceding section on the same page, titled [Connect your devices](https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync#connect-your-devices), states:

[image]

This suggests that I should be able to see both serial numbers when the kinects are externally synchronized, which, as you wrote, is not actually possible. Perhaps it would be a good idea to change the wording of the documentation to indicate this.

rajkundu, 2022-07-16:
There is a section on that same webpage entitled "[Host Computers](https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync#host-computers)" under "Prepare your devices and other hardware":

> ## Host computers
>
> Typically, each Azure Kinect DK uses its own host computer. You can use a dedicated host controller, depending on how you use the device and how much data is transferred over the USB connection.
>
> Make sure that the Azure Kinect Sensor SDK is installed on each host computer. For more information about how to install the Sensor SDK, go to [Quickstart: Set up your Azure Kinect DK](https://docs.microsoft.com/en-us/azure/kinect-dk/set-up-azure-kinect-dk).

This part you are referencing is talking specifically about 3.5 mm cables, which are used for syncing devices and thus are entirely separate from each Azure Kinect's USB connection to its host computer. That is, when it says that the "last device should only have one cable connection", it means "one _[sync]_ cable connection" – because in a daisy-chain configuration, all of the "middle" devices will have two sync cable connections (one in, one out) rather than just one. Below is an illustration from the same webpage:
[image]

I personally think that the current information is sufficient as written, but what do you think?

mdabdk, 2022-07-17:
@rajkundu Looks good to me. Thanks for the help!

closed 2022-07-17 by mdabdk