microsoft/Azure-Kinect-Sensor-SDK #1803: Can I body track a selected area frame?

inigopanos, 2022-07-18:
Hello,

I am trying to make use body tracking to detect and track bodies only in a small area of the screen. What I have is a feed from the kinect, which I modify with background substraction to only have a trapezoid area in which I want to do the body tracking. 

However, I have not been able to just feed that modified video feed to the kinect, and it tracks the bodies using the kinect's default view and FoV, a much larger area than what I want. 

I guess what I want to ask is, is what I want to do even possible, and if so, how can I do it?

I attach an image of what area I want the kinect to body track in (left) and all the camera FoV it tracks in (right)

Thanks

[image]

qm13, 2022-07-25:
Are you segmenting the depth image? Nevertheless body tracking requires approximately half the body in view to work. What is the reasoning behind attempting this technique?

inigopanos, 2022-07-26:
I'm not segmenting the depth image, I'm substracting the background and foreground to try to detect movement on a selected area relative to a static background. 

The reasoning behind this is, I am going to have the Kinect looking at a space with three objects. I want to capture body tracking only in the area adjacent to these obejcts, which is the one on the left (one for each object, so three total), due mostly to performance. 

However, during my tests I have not been able to only 'feed' the selected areas to the body tracking, and have since been trying other approaches. However, I was still wondering if it is possible.