microsoft/Azure-Kinect-Sensor-SDK #1804: Linux compatibility

jiapei100, 2022-07-18:
**Describe the bug**
```console
[  7%] Building C object src/calibration/CMakeFiles/k4a_calibration.dir/calibration.c.o
cd ....../Azure-Kinect-Sensor-SDK/build_22.04/src/calibration && /usr/bin/cc -DARCHITECTURE_x86_64=1 -DFMT_LOCALE -DFMT_SHARED -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -I/usr/local/lib/cmake/inc -I/usr/local/lib/cmake/pal/linux -I/usr/include/libusb-1.0 -I/usr/include/cjson -I....../Azure-Kinect-Sensor-SDK/src/../include -I....../Azure-Kinect-Sensor-SDK/build_22.04/src/sdk/include -D_POSIX_C_SOURCE=200112L  -Werror -Werror -O2 -g -DNDEBUG -fPIC -fvisibility=hidden -Wall -Wextra -Wno-missing-field-initializers -Werror -std=gnu99 -MD -MT src/calibration/CMakeFiles/k4a_calibration.dir/calibration.c.o -MF CMakeFiles/k4a_calibration.dir/calibration.c.o.d -o CMakeFiles/k4a_calibration.dir/calibration.c.o -c ....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c
In file included from ....../Azure-Kinect-Sensor-SDK/src/../include/k4ainternal/handle.h:10,
                 from ....../Azure-Kinect-Sensor-SDK/src/../include/k4ainternal/depth_mcu.h:12,
                 from ....../Azure-Kinect-Sensor-SDK/src/../include/k4ainternal/calibration.h:11,
                 from ....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:5:
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c: In function ‘calibration_create_from_raw’:
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:633:46: error: implicit declaration of function ‘strnlen’ [-Werror=implicit-function-declaration]
  633 |     RETURN_VALUE_IF_ARG(K4A_RESULT_FAILED, !(strnlen(raw_calibration, raw_calibration_size) < raw_calibration_size));
      |                                              ^~~~~~~
....../Azure-Kinect-Sensor-SDK/src/../include/k4ainternal/logging.h:184:10: note: in definition of macro ‘RETURN_VALUE_IF_ARG’
  184 |     if ((_expression_))                                                                                                \
      |          ^~~~~~~~~~~~
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:633:46: error: incompatible implicit declaration of built-in function ‘strnlen’ [-Werror=builtin-declaration-mismatch]
  633 |     RETURN_VALUE_IF_ARG(K4A_RESULT_FAILED, !(strnlen(raw_calibration, raw_calibration_size) < raw_calibration_size));
      |                                              ^~~~~~~
....../Azure-Kinect-Sensor-SDK/src/../include/k4ainternal/logging.h:184:10: note: in definition of macro ‘RETURN_VALUE_IF_ARG’
  184 |     if ((_expression_))                                                                                                \
      |          ^~~~~~~~~~~~
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:656:5: error: unknown type name ‘locale_t’
  656 |     locale_t thread_locale = newlocale(LC_ALL_MASK, "C", (locale_t)0);
      |     ^~~~~~~~
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:656:30: error: implicit declaration of function ‘newlocale’; did you mean ‘setlocale’? [-Werror=implicit-function-declaration]
  656 |     locale_t thread_locale = newlocale(LC_ALL_MASK, "C", (locale_t)0);
      |                              ^~~~~~~~~
      |                              setlocale
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:656:40: error: ‘LC_ALL_MASK’ undeclared (first use in this function)
  656 |     locale_t thread_locale = newlocale(LC_ALL_MASK, "C", (locale_t)0);
      |                                        ^~~~~~~~~~~
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:656:40: note: each undeclared identifier is reported only once for each function it appears in
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:656:59: error: ‘locale_t’ undeclared (first use in this function); did you mean ‘localeconv’?
  656 |     locale_t thread_locale = newlocale(LC_ALL_MASK, "C", (locale_t)0);
      |                                                           ^~~~~~~~
      |                                                           localeconv
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:656:68: error: expected ‘)’ before numeric constant
  656 |     locale_t thread_locale = newlocale(LC_ALL_MASK, "C", (locale_t)0);
      |                                       ~                            ^
      |                                                                    )
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:657:13: error: expected ‘;’ before ‘previous_locale’
  657 |     locale_t previous_locale = uselocale(thread_locale);
      |             ^~~~~~~~~~~~~~~~
      |             ;
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:694:10: error: ‘previous_locale’ undeclared (first use in this function)
  694 |     if ((previous_locale != NULL) && (K4A_FAILED(K4A_RESULT_FROM_BOOL(uselocale(previous_locale) != NULL))))
      |          ^~~~~~~~~~~~~~~
In file included from ....../Azure-Kinect-Sensor-SDK/src/../include/k4ainternal/calibration.h:10,
                 from ....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:5:
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:694:71: error: implicit declaration of function ‘uselocale’; did you mean ‘setlocale’? [-Werror=implicit-function-declaration]
  694 |     if ((previous_locale != NULL) && (K4A_FAILED(K4A_RESULT_FROM_BOOL(uselocale(previous_locale) != NULL))))
      |                                                                       ^~~~~~~~~
....../Azure-Kinect-Sensor-SDK/src/../include/k4a/k4atypes.h:743:34: note: in definition of macro ‘K4A_SUCCEEDED’
  743 | #define K4A_SUCCEEDED(_result_) (_result_ == K4A_RESULT_SUCCEEDED)
      |                                  ^~~~~~~~
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:694:39: note: in expansion of macro ‘K4A_FAILED’
  694 |     if ((previous_locale != NULL) && (K4A_FAILED(K4A_RESULT_FROM_BOOL(uselocale(previous_locale) != NULL))))
      |                                       ^~~~~~~~~~
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:694:50: note: in expansion of macro ‘K4A_RESULT_FROM_BOOL’
  694 |     if ((previous_locale != NULL) && (K4A_FAILED(K4A_RESULT_FROM_BOOL(uselocale(previous_locale) != NULL))))
      |                                                  ^~~~~~~~~~~~~~~~~~~~
....../Azure-Kinect-Sensor-SDK/src/calibration/calibration.c:702:9: error: implicit declaration of function ‘freelocale’; did you mean ‘setlocale’? [-Werror=implicit-function-declaration]
  702 |         freelocale(thread_locale);
      |         ^~~~~~~~~~
      |         setlocale
cc1: all warnings being treated as errors
make[2]: *** [src/calibration/CMakeFiles/k4a_calibration.dir/build.make:79: src/calibration/CMakeFiles/k4a_calibration.dir/calibration.c.o] Error 1
make[2]: Leaving directory '....../Azure-Kinect-Sensor-SDK/build_22.04'
make[1]: *** [CMakeFiles/Makefile2:2641: src/calibration/CMakeFiles/k4a_calibration.dir/all] Error 2
make[1]: Leaving directory '....../Azure-Kinect-Sensor-SDK/build_22.04'
make: *** [Makefile:169: all] Error 2
```

**To Reproduce**
Just build under Linux

**Expected behavior**
Built

**Logs**

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 22.04
 - SDK Version:  1.4.1
 - Firmware version: AzureKinectDK_Fw_1.6.110080014.bin

**Additional context**
N/A

CNCKNAM, 2022-07-18:
The only supported distribution is Ubuntu 18.04

https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download#linux-installation-instructions

dasparli, 2022-07-25:
The current SDK only supports Linux version 18.04

wmcelderry, 2022-08-03:
I've created a repository that builds this project using the docker image (provided by this repository), and extracts the necessary components to run some of the system under later versions of Ubuntu.
See:  [azure_kinect_sdk_scripts](https://github.com/wmcelderry/azure_kinect_sdk_scripts)

It's not perfect, but it may help someone...