microsoft/Azure-Kinect-Sensor-SDK #1805: usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()

jiapei-nexera, 2022-07-19:
Finally, had **Azure-Kinect-Sensor-SDK** built , under **Ubuntu 22.04** .

However, when I tried to run **k4aviewer**, I got the following **ERROR** message:

```console
➜  ~ k4aviewer
[2022-07-19 13:20:06.649] [critical] [t=2705509] ....../Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (365): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2022-07-19 13:20:06.649] [error] [t=2705509] ....../Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2022-07-19 13:20:06.649] [error] [t=2705509] ....../Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
```

Any suggestions?

Eric-chuan, 2023-02-20:
I have met the same issue. Any Solution?

Eric-chuan, 2023-02-20:
Hi, I eventually found out that I had forgotten to copy the [rules.d](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup) to the appropriate location and after doing this step everything worked fine, you can try this.

WentsingNee, 2024-03-17:
> Hi, I eventually found out that I had forgotten to copy the [rules.d](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup) to the appropriate location and after doing this step everything worked fine, you can try this.

Thank you, that helps me A LOT!