microsoft/Azure-Kinect-Sensor-SDK #1810: get body joint coordinates from raw ir and depth images

abhi5454, 2022-07-25:
Hello, i have saved raw depth and IR images from Kinect using a custom application. How i can I extract body joints from these images ?

qm13, 2022-07-25:
Please provide more information.

abhi5454, 2022-07-26:
I am currently using a python application extract and dump IR and depth images in .raw format. I need to run body tracking on those extracted frames and generate corresponding 3d body joint coordinates. How can i create a custom capture object to use my raw inputs and perform bodytracking on the same to extract bodyjoints .
Input - 
1. Kinect IR image (in .raw format)
2. Kinect Depth image (in .raw format)
Output
3D coordinates of Bodyjoints

qm13, 2022-08-08:
What Python wrapper are you using?

abhi5454, 2022-08-09:
I am using PyKinectAzure

qm13, 2022-08-09:
Please direct your question to the developer of PyKinectAzure. You need to combine your .raw format IR and depth images back into a k4a_capture_t structure. You can then enqueue this to the BT queue.