microsoft/Azure-Kinect-Sensor-SDK #1816: RuntimeClassInitialize(). Failed to activate source in open camera: 0x80070005

wangcc123-dave, 2022-08-03:


wangcc123-dave, 2022-08-03:
I am having an issue where the Kinect camera is showing up in the Device Manager, but the Kinect Viewer just says "No available devices." The Kinect Viewer works fine on another computer, so is not defective.

Windows 10

dasparli, 2022-08-08:
Are you only seeing one Kinect camera in the device manager? You should be seeing two, one called Azure Kinect 4K Camera and the other called Azure Kinect Depth Camera.

Do you see any differences in the device managers between the working and non-working computer?

Is this failure new on the computer that is not working? Perhaps you may have luck with uninstalling and reinstalling the AKDK software.