microsoft/Azure-Kinect-Sensor-SDK #1818: Depth sensor flickering in imseCAVE

gfsd20433, 2022-08-08:
When the Azure Kinect is recording in the imseCAVE, the depth sensor keeps flickering, whenever in a steady environment.

Here is the example [](https://youtu.be/h5wXIvYpunk
)

I suppose I can take a depth sensor snapshot for background evironment and checks if people (or any other things) enters the imseCave. If the result keeps flickering, there will be wrong result.

Any possible fix?
Thanks in advance.

dasparli, 2022-08-08:
If the flickering you are talking about is occurring at the 9 second mark in the lower left area of the depth image, that is probably caused by a speck of dust floating by the sensor. The ToF sensor is really good at picking up dust in the air.

gfsd20433, 2022-08-08:
The dust problem is reasonable to me because the infrared camera also recorded the dust. 

I was talking about the situation like 0:13/0:17/0:33 in the depth image. You can see there seems nothing changed in the color camera (I dont think the projection effects the infrared camera), but part of the depth camera image suddenly goes black (no signal) a second and then return to normal for unknown reason.

And the flickering problem actually occurs throughout the video. The above 3 time marks are the most obvious we can see.

I have tried to use the Kinect in other environment and I think it is normal to have a few pixels unstable. I expect the normal result will be like 0:25~0:27 for the whole video. But it goes unpredictable when I use it in imseCAVE.

dasparli, 2022-08-11:
This looks a lot like interference from another IR light source. And it looks like it is happening in a repeating pattern every 2 or 3 seconds. Do you have another device in the room that is emitting IR light around 850nm?

closed 2022-08-22 by dasparli