microsoft/Azure-Kinect-Sensor-SDK #1819: "Error: 0:2(38): error: `gl_FragColor' undeclared" in ubuntu20.04

judgeee-wase, 2022-08-16:
****

**Describe the bug**

 I tried to use the ”Body tracking SDK” with AzurekinnectDK in Ubuntu20.04. However, after the SDK starts up halfway through, a window appears for a moment, then it crashes immediately and gives me an error message. I am a beginner in using ubuntu. In addition, my English is poor, so I will give details in English as much as possible, but if you still have questions, please ask. 

**To Reproduce**

1. REGISTER THE REPOGITORY&INSTALLING “SensorSDK” 
    I do the following command .

   `curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo tee /etc/apt/trusted.gpg.d/microsoft.asc `
   
   `sudo apt-add-repository https://packages.microsoft.com/ubuntu/20.04/prod` 
   
   `sudo apt-get update`

   And then,the following statement will be added to the bottom of the text in /etc/apt/sources.list/. I figured I could now register the Microsoft repository. 

   

   > deb [arch=amd64] https://packages.microsoft.com/ubuntu/20.04/prod focal main 
   > #deb-src [arch=amd64] https://packages.microsoft.com/ubuntu/20.04/prod focal main 

   And, I do
   `sudo apt install libk4a1.4dev `
   `sudo apt install k4a-tools`

   I suceeded in installing “sensorSDK”. And I suceeded in using "k4aviewer"
   by doing the command.

   `sudo k4aviewer`

2. INSTALLING CUDA & CUDnn 
   Next, CUDA and CUDnn were installed. I installed these versions from a friend 
   who had used BodyTrackingSDK in a ubuntu 20.04 environment. 

3. INSTALLING “BodyTracking SDK” & USING 
   `sudo apt install libk4abt1.0-dev`

   Finally, in a terminal, with sudo privileges, run "sudo k4abt_simple_3d_viewer". 
   It crashes after displaying

   >    `Basic Navigation:
Rotate: Rotate the camera by moving the mouse while holding mouse left button
    Pan: Translate the scene by holding Ctrl key and drag the scene with mouse left button
    Zoom in/out: Move closer/farther away from the scene center by scrolling the 
   mouse scroll wheel
    Select Center: Center the scene based on a detected joint by right clicking t 
  he joint with mouse
 Key Shortcuts
 ESC: quit
 h: help
 b: body visualization mode
 k: 3d window layout

   and displays 

   > Shader Error: 0:2(38): error: `gl_FragColor' undeclared
   > 0:2(38): error: value of type vec4 cannot be assigned to variable of type error

 is displayed. **

**Expected behavior**

  I thought I can use Body Tracking SDK.

**Logs**

   > Shader Error: 0:2(38): error: `gl_FragColor' undeclared
   > 0:2(38): error: value of type vec4 cannot be assigned to variable of type error

**Screenshots**

**Desktop (please complete the following information):**
 - Ubuntu20.04LTS: (Now processer: AMD® Ryzen 5 4600h with radeon graphics × 12 )
 - CUDnn : 1.0-1 
 - CUDA SDK version: 11.7.1
 - Graphics:NVIDIA Corporation / AMD® Renoir -> Geforce1650 Super  /

**Additional context**

The code & the results are as follows.
`glxinfo | grep OpenGL`

> OpenGL vendor string: AMD
OpenGL renderer string: AMD RENOIR (DRM 3.42.0, 5.15.0-46-generic, LLVM 12.0.0)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.2.6
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.2.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
core profile version string: 4.6 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.2.6
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.2.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

`nvidia-smi`

> +-----------------------------------------------------------------------------+
> | NVIDIA-SMI 515.65.01    Driver Version: 515.65.01    CUDA Version: 11.7     |
> |-------------------------------+----------------------+----------------------+
> | GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
> | Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
> |                               |                      |               MIG M. |
> |===============================+======================+======================|
> |   0  NVIDIA GeForce ...  On   | 00000000:01:00.0 Off |                  N/A |
> | N/A   45C    P8     5W /  N/A |     10MiB /  4096MiB |      0%      Default |
> |                               |                      |                  N/A |
> +-------------------------------+----------------------+----------------------+
>                                                                                
> +-----------------------------------------------------------------------------+
> | Processes:                                                                  |
> |  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
> |        ID   ID                                                   Usage      |
> |=============================================================================|
> |    0   N/A  N/A      1047      G   /usr/lib/xorg/Xorg                  4MiB |
> |    0   N/A  N/A      1720      G   /usr/lib/xorg/Xorg                  4MiB |
> +-----------------------------------------------------------------------------+
> 

`lspci | grep VGA`
> 01:00.0 VGA compatible controller: NVIDIA Corporation Device 2192 (rev a1)
> 05:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c7)

qm13, 2022-08-23:
This is a known issued fixed in Body Tracking SDK v1.1.2. Your package should be libk4abt1.**1**-dev.

judgeee-wase, 2022-08-29:
@qm13 
Thank you for your advice! I succeeded in using Body Tracking SDK in Ubuntsu20.04 !

First ,I used MS repository for 20.04, but it is not good.
I change MS repository for 18.04 instead of using 20.04.
`sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/prod
`
And second , I installed k4abt=1.3 .and I tried to upgrade this(to version 1.4)
But I cannnot overwrite .so file. So, I used
`sudo dpkg --force-all -i  `

and this enabled me to use SDK.
I upgraded v1.1.2.

Really thank you!!!!
I'm going to close this Comment.

closed 2022-08-29 by judgeee-wase