microsoft/Azure-Kinect-Sensor-SDK #1824: Issues closed with wrong info, missing info, or prematurely with no followthrough or tracking

diablodale, 2022-08-31:
The following issues are missing information, have wrong information, or prematurely closed with no followthrough, tracking, or responses. This is again repeated poor Microsoft behavior.  It is a lack of adequate professional levels of quality code releases, communication, issue management, PR management, and documentation.

I escalated on 2 Mar 2020 to your GPM Neena Kamath due to similar failures. I will again escalate to them or the replacement GPM.

I limit discussion here to only this subset of issue management failures...

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1782
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1576
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1573
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1546
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1541
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1373
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1365
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1607
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1540
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1414

qm13, 2022-09-12:
@diablodale I have updated most of the issues you list. Please reactivate the ones you think require further information.

diablodale, 2022-09-13:
@qm13, it is not possible for customers like me to "reactivate" issues. Your team did not grant github permissions to do so, and comments on closed issues are ignored by your team (which lead to my escalation and this issue).

Please adjust or update me on the process to "reactivate" issues.

jalex8188, 2022-09-22:
Bumping this issue with a comment, as I am eager to utilize @diablodale 's MaxMSP object for the azure kinect