microsoft/Azure-Kinect-Sensor-SDK #1828: k4arecord lib is broken, fails to initialise due to spdlog exception

eidetic-av, 2022-09-06:
**Describe the bug**

Projects using k4arecord lib crash at runtime due to a spdlog exception claiming the k4a_logger is being initialised twice:
```
terminate called after throwing an instance of 'spdlog::spdlog_ex'
  what(): logger with name 'k4a_logger' already exists
```

In C++, the following simple code throws the above error:
``` C++
#include <iostream>
#include <k4arecord/playback.h>

int main() {
  auto playback = k4a::playback::open("any-recording.mkv");

  k4a::capture capture;
  playback.get_next_capture(&capture);
  std::cout << "size: " << capture.get_color_image().get_size() << "\n";

  return 0;
}

```
Also, I have attempted to build and run the C example at [examples/playback_external_sync](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/playback_external_sync) which again fails with the same exception.

**To Reproduce**

1. Install SDK
2. Build either examples above
3. Notice exception thrown at runtime

**Expected behavior**

No exception. Logger is initialised only once.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 
- OS: Arch Linux
- SDK Version: 1.4.1, also tried compiling latest in develop branch

**Additional context**

dasparli, 2022-09-29:
The k4a.lib and k4arecord.lib can't log to the same file. There is a note about this in the documentation: [Azure Kinect Sensor SDK: Error and event logging (microsoft.github.io)](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___logging.html)

Confirm that your environment variables are using the same log file name for both libs.

eidetic-av, 2022-10-04:
@dasparli thanks for this info. I'll try this again but set these environment variables. I wasn't aware of their existence.