microsoft/Azure-Kinect-Sensor-SDK #1833: Camera device not opened on a build of the sample_unity_bodytracking

Thebinoman, 2022-09-19:
When running a build of the [sample project for Body Tracking on Unity](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking) It seems that the camera device does not open. Using Unity Debugger on Visual Studio, I can see it fails on line 32 on SkeletalTrackingProvider.cs:
`using (Device device = Device.Open(id))`

I followed the instructions on this thread: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1728
and this video: https://youtu.be/PGsxP6Yoq9I
and It works O.K. on the Unity Editor.
Then I saw it doesn't work when I build it into a stand-alone build. I followed this thread: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1767 and this thread: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1684
and copied the mentioned files into the same folder of the UnityCSharp.exe file, but nothing changed.

**To Reproduce**
1. Install Azure Kinect Sensor SDK
2. Install Azure Kinect Body Tracking SDK
3. Download the body tracking sample for Unity from https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking
4. Follow the instruction on this video: https://youtu.be/PGsxP6Yoq9I
5. Build a stand-alone app from within Unity.
6. Follow [this thread](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1684) and copy the files into the same folder of UnityCSharp.exe.
7. Run UnityCSharp.exe

Tested on Windows 10 with RTX 1070

amantley, 2022-10-11:
Hi, are  you building for any or all of GPU CPU CUDA or DirectML?  Also what version of Cuda software do you have?  I am not have a problem with cuda and cudnn 11.4 
[image]

corriedotdev, 2023-05-03:
Ensure you copy the relevant dlls. Used dependancy walker to test and here is what i think you need to put in the root of the exe as well after the build is complete.. probs best to put this into a bat or something. 

In the current dir import 
```
cublas64_11.dll
cublasLt64_11.dll
cudart64_110.dll
cudnn64_8.dll
cudnn_cnn_infer64_8.dll
cudnn_ops_infer64_8.dll
cufft64_10.dll
dnn_model_2_0_op11.onnx
onnxruntime.dll

```