microsoft/Azure-Kinect-Sensor-SDK #1838: I'd like to create a variant of the white cover that would allow me to mount other sensors on the Kinect. Does anyone know if there is a 3D printable model file of the white cover that I could use as the basis to modify?

DrMaxP, 2022-09-25:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

dasparli, 2022-10-03:
There is a link to a STEP file that includes this information which you can find on this page: https://learn.microsoft.com/en-us/azure/kinect-dk/hardware-specification#product-dimensions-and-weight

DrMaxP, 2022-10-08:
That could be useful, but it gives the dimensions of what is under the white cover, not the cover itself - although most dimensions can be inferred from this file.

rajkundu, 2022-10-29:
I made a "double" camera mount a little while back on Fusion 360 that I would be happy to share. I made it by painstakingly following that STEP file to create a housing that fits around the "bare" Kinect DK. I'm not the best at CAD, however, so my model isn't exactly professional-grade. Perhaps it could be a starting point, though, if you're interested?

DrMaxP, 2022-11-14:
Sounds great! Yes please. I'm m I've started procedurally generating a model using code, but is slow work. I have the shell but none of the holes in it.

rajkundu, 2022-11-15:
I found the "single" camera version of my mount. Here it is: [AKDK Sleeve.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/10009126/AKDK.Sleeve.zip).

Let me know if I can provide you with anything else! If you are also using Fusion 360, then perhaps we can find a way to more natively share the design so that you can easily alter the model to suit your needs.