microsoft/Azure-Kinect-Sensor-SDK #1841: Azure Kinect device detected by windows but not found in the viewer and the other K4a apps

frenchtouchXR, 2022-09-27:
I had my devices working fine on my laptop untill today. They are no more usable by the K4a apps and libraries. I get the error : 
[ error    ] : usb_cmd_get_device_count(). No devices found in the azure viewer. The cameras work fine on my other computer. I changed the cables, double checked the alimentation (it's solid white), changed the usb 3.0 plugs and restarted 20 time the compute. Still not found. 
 
[image]
[image]

dasparli, 2022-09-28:
Did anything on your laptop change between the last time it worked and today?

Have you tried running another USB device on the same connectors that your Kinect was connected to?

frenchtouchXR, 2022-09-28:
Hi, Maybe a windows update. I noticed my laptop's bisoncam isn't found anymore by device manager. I don't know if it has something to do.

frenchtouchXR, 2022-09-28:
In the device manager of windows 10 I deactivated/reactivated the webcam : Azure Kinect 4K camera. It then was found again by the kinect viewer. I can't explain what happened.

closed 2022-09-28 by frenchtouchXR