microsoft/Azure-Kinect-Sensor-SDK #1842: Unreliable confidence levels of joints

jonathansim94, 2022-09-28:
We are working with the latest versions of the body tracking SDK. Currently you provide 3 confidence levels (from None to Medium) and plan to implement the fourth (High) in future SDKs.

We are working in a multi kinect environment (reproducing in real time an acquired movement on an avatar in a Unity scene) and we have noticed a reliability problem about these confidence values. In particular, we merge the positions of the different avatar's bones captured by the different kinects and even when a bone is not visible from a sensor the maximum confidence level ("Medium") is sent for that joint with an inferred rotation. Due to this, the merge algorithm uses correct and totally wrong data and the affected bones assume wrong positions.

This issue, therefore, wants to highlight a problem on the unreliability of the confidence level whose correctness is crucial for our purposes. Currently we have mitigated with an algorithm that tries to understand which kinect sends the wrong data using a buffer, but this obviously cause an evident delay between the actual acquisition of data from the sensors and the movement of the avatar.

Are you planning to improve the reliability of these confidence levels or to release soon an implementation for the "High" confidence level which I assume will guarantee a better reliability?

qm13, 2022-10-03:
The confidence level is driven by what the AI returns (NOT what the camera sees). Consequently, the AI may return a joint position for a joint not in view of the camera e.g., hidden under the table, and vice versa.

tiborboglar, 2022-10-05:
I also found the confidence levels to be unreliable using the latest version of the SDK, improving these scores would be really nice for multi Kinect applications. Is there a date for the release of the "High" confidence joint pose estimation? Also, why not transform it to a [0, 1] range instead of using confidence bands?