microsoft/Azure-Kinect-Sensor-SDK #1843: How to set direction in Unity 3d Azure kinect Bodytracking?

ghkd3930, 2022-10-05:
I'm following the Unity BodyTracking sample, but it's frustrating because the modeling is always looking behind me.
Changing or angled the matched bones will break or reverse the behavior.
Is there any way to change the viewing direction within Azure kinect itself?

ghkd3930, 2022-10-12:
> 안녕하세요, 여기 이 코드에 있는 것처럼 오프셋 회전을 추가할 수 있습니다. 그런 다음 관리자에서 회전을 변경할 수 있습니다. [image] [image]

hello. Thank you for answer. The method you provided is no different than adjusting the Y_180_FLIP of the TrackerHandler. If I raise my left hand, that friend also turns upside down and raises my left hand so that the mirror doesn't come out. I want a mirror-like picture.

Lt23live, 2024-03-26:
I know this was a while ago but did you ever figure out how to solve it?